realsenseai/librealsense #5723: "Frames didn't arrived within 5 seconds" on Ubuntu 18.04.3 (Kernel 5.3.0-26) with librealsense-2.32.1

byeongkyu, 2020-01-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |  Linux (Ubuntu 18.04.3  | 
| Kernel Version (Linux Only)    | 5.3.0-26-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  librealsense-2.32.1                          | 
| Language                            |                            | 
| Segment			|                           | 

### Issue Description

When I check the depth and rgb image using the tool (realsense-viewer, rs-align, etc.), first time is looking good. But after turn off and on again, it is not working until hardware reset or plug out and in.

The error message is "Timestamp: 1580194238836.964600
Severity: Warn
Description: Frames didn't arrived within 5 seconds"

I have checked the installation using binary or source build. Both cases have same problem.

ev-mp, 2020-02-02:
@byeongkyu, please attach the kernel and application logs for review.

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@byeongkyu Could you please provide more information as ev-mp commented? Thanks!

byeongkyu, 2020-02-12:
There are no special messages.

 12/02 15:34:15,511 WARNING [140170099660544] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:20,515 WARNING [140170099660544] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:29,856 WARNING [140170108053248] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:30,687 WARNING [140170082875136] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:34,861 WARNING [140170108053248] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:35,690 WARNING [140170082875136] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds
 12/02 15:34:39,866 WARNING [140170108053248] (backend-v4l2.cpp:973) Frames didn't arrived within 5 seconds

dmesg

[   29.216371] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.370123] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.421374] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.475126] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.561374] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.612632] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   29.694874] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

[image]

RealSenseCustomerSupport, 2020-02-18:
----------------------------------------------

@byeongkyu Is it possible for you to try other kernel version such as kernel 5.0 to see if have the same issue?

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@byeongkyu Did you get chance to try kernel 5.0? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@byeongkyu According to the release note of SDK v2.33.1, there's some fix for kernel 5.3. Please have a try with the latest SDK to see if you still have issue. Thanks!
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2331

byeongkyu, 2020-03-22:
@RealSenseCustomerSupport Sorry for late answer. But it still has an error with kernel 5.3 and SDK v2.33.1. 

Kernel: 5.3.0-42-generic #34~18.04.1-Ubuntu SMP
SDK: librealsense 2.33.1-0~realsense0.2079

RealSenseCustomerSupport, 2020-03-24:
----------------------------------------------

@byeongkyu What's the firmware version of your camera? I didn't get such issue on my setup with the latest firmware and SDK v2.33.1 on Ubuntu 18.04 kernel 5.3.0-40. If you're using old firmware, could you please try to upgrade it to latest one to see if it helps? Thanks!

byeongkyu, 2020-03-24:
@RealSenseCustomerSupport Of course I installed latest firmware using realsense-viewer. I will check on the other computer. Thanks for your reply.

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@byeongkyu Thanks for the update! Looking forward to your test update on other computer.

byeongkyu, 2020-04-01:
@RealSenseCustomerSupport  On my laptop (kernel 5.4.27+), it works successfully. I will check my desktop. Thanks for your help. :-)

closed 2020-04-01 by byeongkyu