realsenseai/librealsense #5725: [E] TrackingManager: Error while loading file to device 0x0x7f8401ebb0 [T265]

zainmehdi, 2020-01-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Operating System & Version |   Linux- Ubuntu Bioninc 18.01 |                              
| Platform                                 | NanoPi Neo4 |
| SDK Version                            |  { 2.31 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am developing on an embedded board Nano Pi Neo 4 using QT creator. When I try to debug my application in Debug mode I am unable to connect to T265 and get this error

[E] TrackingManager: Error while loading file to device 0x0x7f8401ebb0
When I switch back to Release mode this error doesnt appear any more.  Can someone please provide an hints .Thanks

zainmehdi, 2020-02-07:
Hi @radfordi , can you please provide some hints on hit. I am unable to run my program in debug mode because of this error.

radfordi, 2020-02-07:
Can you extract from your application a simple example program that shows the issue?

zainmehdi, 2020-02-10:
Its seems like its more of a QT error than librealsense. i am still trying to figure out and update asap.

closed 2020-11-25 by zainmehdi