realsenseai/librealsense #5726: RealSense-Matlab: multiple device

neripaolo, 2020-01-28:
Dear users

I'm new in realsense-matlab interface. I tryed the simple example to acquire point clouds, and they worked fine so far. Now I want to use more than one sensor, and I want to chose which one to use for each acquisition (timing is not an issue so far). How can I select which device to stream for in matlab? (I'm using D415). 

Thank you for your help!

This is the code I'm using to start treaming from arbitrary device:
  pipe = realsense.pipeline();
    
    pointcloud = realsense.pointcloud();
%%
    % Start streaming on an arbitrary camera with default settings
    profile = pipe.start();

    figure('visible','on');  hold on;
    figure('units','normalized','outerposition',[0 0 1 1])

        CLOUDS={};
        nc=40;
    % Main loop
    for i = 1:nc

        % Obtain frames from a streaming device
        fs = pipe.wait_for_frames();
        
        % Select depth frame
        depth = fs.get_depth_frame();
        %color = fs.get_color_frame();

        
        % Produce pointcloud
        if (depth.logical())% && color.logical())

            %pointcloud.map_to(color);
            points = pointcloud.calculate(depth);
            
            % Adjust frame CS to matlab CS
            vertices = points.get_vertices();
            X = vertices(:,1,1);
            Y = vertices(:,2,1);
            Z = vertices(:,3,1);

            plot3(X,Z,-Y,'.');
            grid on
            hold off;
            view([45 30]);

%             xlim([-0.5 0.5])
%             ylim([0.3 1])
%             zlim([-0.5 0.5])

            xlabel('X');
            ylabel('Z');
            zlabel('Y');
            
            pause(0.5);
            
            
        CLOUDS{i}=[X Z -Y]*1000;
        end
        % pcshow(vertices); Toolbox required
    end

     % Stop streaming
    pipe.stop();

lramati, 2020-02-25:
take a look at the config object. You can pass it to pipeline.start to request a specific device / specific streams and settings.
e.g.
```matlab
cfg = realsense.config();
cfg.enable_device('<serial number>');
cfg.enable_stream(realsense.stream.depth, 1280, 720, realsense.format.z16);
pipe.start(cfg);
```

neripaolo, 2020-02-26:
Thank you for your answer, it solved the problem!
Now I'm facing another issue, with SR300 in a multi-sensor setup: I want to switch them on and of sequently, to avoid interference between the sensors' field of view. So far i managed to do so by changing the laser power from 0 to max. Anyway, this procedure results slow, since a pause is needed to allow the sensors to set after the laser power is changed: Is there a faster way to control this simulatneous acquisition of the same scene without sensors cross-talk?

Thank you very much!

closed 2020-12-14 by RealSenseSupport