realsenseai/librealsense #5729: Set the depth camera properties in c++

kevinpaulsudarshan, 2020-01-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   {Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {computer vision}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I am new to the realsense SDK. I am trying to modify the depth properties(viz. DEPTH_PROJECTOR_POWER, DEPTH_MOTION_RANGE, etc.) of the Intel realsense SR300 in c++. I want the similar behaviour as in the realsense viewer where I can drag the trackbar and change motion to range trade off, etc. Can you please give me a code snippet that performs the same. Thank you in advance for your help and answers.

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@kevinpaulsudarshan The source code of RealSenseViewer is in the link below. Please check if this helps you. Thanks!
https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer

kevinpaulsudarshan, 2020-02-07:
Thanks for your answer @RealSenseCustomerSupport. Actually I found the answer to my question in the sensor-control [example.](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control). Here is a small code snippet if anybody needs in the future. 
    
    rs2::context ctx; 
    rs2::device_list devices = ctx.query_devices();
    rs2::device selected_device;
    
    if (devices.size() == 0)
    {
            std::cerr << "No device connected, please connect a RealSense device" << std::endl;

            //To help with the boilerplate code of waiting for a device to connect
            //The SDK provides the rs2::device_hub class
            rs2::device_hub device_hub(ctx);

            //Using the device_hub we can block the program until a device connects
            selected_device = device_hub.wait_for_device();
    }
    else
    {
        selected_device = devices[0]; //select device 0 as there is only one device connected in my case
    }
    std::vector<rs2::sensor> sensors = selected_device.query_sensors();
    rs2::sensor sensor = sensors[0]; //select sensor 0 as it is the depth sensor in my case
   
    // Each option provides its rs2::option_range to provide information on how it can be changed
    // To get the supported range of an option we do the following:
    rs2_option option_type = RS2_OPTION_MOTION_RANGE; //using depth motion range trade-off as the option
    float newValue = 100;
    rs2::option_range range = sensor.get_option_range(option_type);
    float default_value = range.def;
    float maximum_supported_value = range.max;
    float minimum_supported_value = range.min;
    float difference_to_next_value = range.step;

    if(minimum_supported_value <= newValue <= maximum_supported_value)
    {
        sensor.set_option(option_type, newValue);
    }
    else
    {
        std::cout << "  Input value of option type is out of range! " << std::endl;
        std::cout << "  Supported range for option " << option_type << ":" << std::endl;
        std::cout << "  Min Value     : " << minimum_supported_value << std::endl;
        std::cout << "  Max Value     : " << maximum_supported_value << std::endl;
    }

closed 2020-02-07 by kevinpaulsudarshan