realsenseai/librealsense #5730: OpenCV RGB-D Odometry

raymanfx, 2020-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |  Linux (Raspbian Buster)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.19.93)                                         | 
| Platform                                 | Raspberry Pi 4B  |
| SDK Version                            |  GitHub master                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                         | 

### Issue Description
First of all, thanks for the fantastic library. It's been a pleasure to work with so far and my only issues have been related to Linux USB flakiness in general.

I'm working on a project where I have a robot equipped with the D435 camera. The robot is controlled by an external PC. I need to stream RGB and Depth data in lockstep to the PC over WiFi.
I set up an Archer C6v2 5Ghz AP and I'm getting ~90 Mbits/sec in either direction, so that is not the limiting factor for me.

On the software side, I want to use the RGB and Depth data to compute the visual odometry using OpenCV. The streaming pipeline is established and works well so far. I also built in JPEG compression for the RGB frames and ZSTD for the 16 bit grayscale ones, which is all working nicely.

The OpenCV RGBD module doesn't seem to be documented overly well (https://docs.opencv.org/4.1.0/df/ddc/classcv_1_1rgbd_1_1Odometry.html and https://docs.opencv.org/4.1.0/d0/d60/classcv_1_1rgbd_1_1RgbdOdometry.html). From what I can see, I need a camera matrix for this to work. I can easily get the Cx,Cy Fx,Fy values from the RealSense tools, but I'm not sure whether I need to fill in the depth or color sensor intrinsics here?
Furthermore, it looks like I cannot do real-time image alignment (depth-to-color or color-to-depth) on the Raspberry, it's just too slow. Now I can probably do it on the host machine instead using the RealSense provided extrinsics, but I'm not sure where I would need to do that.
If I rectify/align the images, why do I need to pass the camera matrix to OpenCV?

I apologize if this is more of an OpenCV question rather than a RealSense one. Please let me know if you are able to help me with this issue or if I should go ask somewhere else.

TL;DR
- Which camera matrix is needed to compute visual odometry using OpenCV?
- What is the best way to align the images on the external PC?
- Are there any pre-made examples of visual odometry using RealSense?

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@raymanfx Some OpenCV examples are in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv  Please check if it helps. Thanks!

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@raymanfx  Any other questions about this? Looking forward to your update. Thanks!

raymanfx, 2020-04-14:
Hi, your OpenCV examples were not helpful obviously.
In the meantime, I figured out how to align the depth to the color frame on my host machine while streaming the images to from the Raspberry Pi.

Leaving out my custom solution for streaming compressed color and depth data in realtime, I basically did the following:
```
<RemoteCapture> cap;
```

Get the depth -> color extrinsics from the realsense pipeline profiles:
```
    auto color_profile = cap.pipeline().get_active_profile().get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
    auto depth_profile = cap.pipeline().get_active_profile().get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
    auto depth2color_extrin = depth_profile.get_extrinsics_to(color_profile);
```

Create the OpenCV camera matrix (intrinsics):
```
    // camera matrix layout:
    // fx  0  cx
    // 0   fy cy
    // 0   0  1

    // rs-enumerate-devices -c
    // color intrinsics
    cv::Mat color_intrinsics(3, 3, CV_32FC1);
    color_intrinsics.at<float>(0, 0) = color_profile.get_intrinsics().fx;
    color_intrinsics.at<float>(0, 1) = 0;
    color_intrinsics.at<float>(0, 2) = color_profile.get_intrinsics().ppx;
    color_intrinsics.at<float>(1, 0) = 0;
    color_intrinsics.at<float>(1, 1) = color_profile.get_intrinsics().fy;
    color_intrinsics.at<float>(1, 2) = color_profile.get_intrinsics().ppy;
    color_intrinsics.at<float>(2, 0) = 0;
    color_intrinsics.at<float>(2, 1) = 0;
    color_intrinsics.at<float>(2, 2) = 1;

    // rs-enumerate-devices -c
    // color intrinsics
    cv::Mat depth_intrinsics(3, 3, CV_32FC1);
    depth_intrinsics.at<float>(0, 0) = depth_profile.get_intrinsics().fx;
    depth_intrinsics.at<float>(0, 1) = 0;
    depth_intrinsics.at<float>(0, 2) = depth_profile.get_intrinsics().ppx;
    depth_intrinsics.at<float>(1, 0) = 0;
    depth_intrinsics.at<float>(1, 1) = depth_profile.get_intrinsics().fy;
    depth_intrinsics.at<float>(1, 2) = depth_profile.get_intrinsics().ppy;
    depth_intrinsics.at<float>(2, 0) = 0;
    depth_intrinsics.at<float>(2, 1) = 0;
    depth_intrinsics.at<float>(2, 2) = 1;
```

Account for rotation and translation to match the camera lens positions (extrinsics):
```
    // rs-enumerate-devices -c
    // extrinsics (depth to color)
    cv::Mat extrin_rot(3, 3, CV_32FC1);
    cv::Mat extrin_trans(1, 3, CV_32FC1);
    extrin_rot.at<float>(0, 0) = depth2color_extrin.rotation[0];
    extrin_rot.at<float>(0, 1) = depth2color_extrin.rotation[3];
    extrin_rot.at<float>(0, 2) = depth2color_extrin.rotation[6];
    extrin_rot.at<float>(1, 0) = depth2color_extrin.rotation[1];
    extrin_rot.at<float>(1, 1) = depth2color_extrin.rotation[4];
    extrin_rot.at<float>(1, 2) = depth2color_extrin.rotation[7];
    extrin_rot.at<float>(2, 0) = depth2color_extrin.rotation[2];
    extrin_rot.at<float>(2, 1) = depth2color_extrin.rotation[5];
    extrin_rot.at<float>(2, 2) = depth2color_extrin.rotation[8];
    extrin_trans.at<float>(0, 0) = depth2color_extrin.translation[0];
    extrin_trans.at<float>(0, 1) = depth2color_extrin.translation[1];
    extrin_trans.at<float>(0, 2) = depth2color_extrin.translation[2];
```

Assemble a final rigid body transform matrix:
```
    // rigid body transform
    // https://en.wikipedia.org/wiki/Camera_resectioning#Extrinsic_parameters
    cv::Mat Rt(4, 4, CV_32FC1);
    Rt.at<float>(0, 0) = extrin_rot.at<float>(0, 0);
    Rt.at<float>(0, 1) = extrin_rot.at<float>(0, 1);
    Rt.at<float>(0, 2) = extrin_rot.at<float>(0, 2);
    Rt.at<float>(0, 3) = extrin_trans.at<float>(0, 0);
    Rt.at<float>(1, 0) = extrin_rot.at<float>(1, 0);
    Rt.at<float>(1, 1) = extrin_rot.at<float>(1, 1);
    Rt.at<float>(1, 2) = extrin_rot.at<float>(1, 2);
    Rt.at<float>(1, 3) = extrin_trans.at<float>(0, 1);
    Rt.at<float>(2, 0) = extrin_rot.at<float>(2, 0);
    Rt.at<float>(2, 1) = extrin_rot.at<float>(2, 1);
    Rt.at<float>(2, 2) = extrin_rot.at<float>(2, 2);
    Rt.at<float>(2, 3) = extrin_trans.at<float>(0, 2);
    Rt.at<float>(3, 0) = 0;
    Rt.at<float>(3, 1) = 0;
    Rt.at<float>(3, 2) = 0;
    Rt.at<float>(3, 3) = 1;
```

Finally, align depth and color images using OpenCV:
```
cv::Mat frame_rgb;
cv::Mat frame_depth;
cv::Mat frame_depth_reg;

/* initialize frames from remote source */

cv::rgbd::registerDepth(depth_intrinsics, color_intrinsics, cv::Mat(), Rt, frame_depth,
                                    frame_rgb.size(), frame_depth_reg, true /* dilate */);
```

I hope this helps the next person wondering how to do this.
If you want, I can create a pull request with a stripped down example that needs just librealsense and OpenCV.

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@raymanfx Thanks very much for your sharing! Pull requests are always welcome.

raymanfx, 2020-04-15:
I created a pull request: https://github.com/IntelRealSense/librealsense/pull/6242.

RealSenseCustomerSupport, 2020-04-16:
----------------------------------------------

@raymanfx Thanks very much for your contribution!

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@raymanfx Our engineering team is looking at the PR https://github.com/IntelRealSense/librealsense/pull/6242 Let's track there and close this accordingly. Thanks!

closed 2020-05-18 by RealSenseCustomerSupport