realsenseai/librealsense #5735: 2.32.1 regression with set_motion_device_intrinsics()

JBBee, 2020-01-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.32.1                        | 
| Language                            |  C++                          | 
| Segment			|  AR                        | 

### Issue Description
The code below works with 2.31.0 but with the latest it throws: `RealSense error calling rs2_set_motion_device_intrinsics(sensor:00000180FA8AB1B0, profile:00000180F8E34828, intrinsics:0000007736F3FD70):
    Invalid stream index`

```cpp - C++
rs2::context ctx;
rs2_motion_device_intrinsic read_motion_intrinsics[2] = {};
for (auto dev : ctx.query_devices())
{
    if (auto tm2 = dev.as<rs2::tm2>())
    {
        for (auto sensor : tm2.query_sensors())
        {
            for (auto profile : sensor.get_stream_profiles())
            {
                if (auto mp = profile.as<rs2::motion_stream_profile>())
                {
                    read_motion_intrinsics[mp.stream_type() == RS2_STREAM_GYRO ? 0 : 1] = mp.get_motion_intrinsics();
                }
            }
        }

        for (auto imu : { RS2_STREAM_GYRO, RS2_STREAM_ACCEL })
        {
            //std::cout << "Going to write motion intrinsics: " << std::endl;
            //print(read_motion_intrinsics[imu - RS2_STREAM_GYRO]);
            tm2.set_motion_device_intrinsics(imu, read_motion_intrinsics[imu - RS2_STREAM_GYRO]);
        }
        tm2.write_calibration();
        std::cout << "Set motion device intrinsics" << std::endl;
    }
}
```

Thanks.

RealSenseSupport, 2021-03-24:
Hi,

Will you be needing further help with this?

*If we don’t hear from you in 7 days, this issue will be closed.*

Thanks

JBBee, 2021-03-24:
Hi,

Thanks for checking.  This isn't something that's currently holding us up so I will close it.  I'll open a new issue if it comes up again.

closed 2021-03-24 by JBBee