realsenseai/librealsense #5736: get_gvd_field() implementation

JTrantow, 2020-01-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (NA) | 
| Operating System & Version |   {Win10| 
| Platform                                 | PC|
| SDK Version                            |  2.0 master| 
| Language                            |  {C++ }                          | 

### Issue Description
get_gvd_field() has a couple problems.
1) When the return type is bool, the code may work but it triggers a C4804 compiler warning since the '+=' isn't safe for type bool.

2) The test against the data.size() is incorrect.
                if (index + sizeof(T) >= data.size())
It should throw only on the '>'. This function as is will fail when trying to return any type from the last data byte.

The only call to this function I see is in I500-device.cpp where it checks the camera lock where it does not use the last data byte, but is used with a bool.

MartyG-RealSense, 2020-03-06:
Apologies for the delay in receiving a response to this question.

There was a failed check on the pull that you created, so I would recommend taking a look at this issue if you still want that pull to go ahead.   Thanks!

https://github.com/IntelRealSense/librealsense/pull/5737

MartyG-RealSense, 2020-03-12:
Case closed due to no further responses.

JTrantow, 2020-10-28:
Please reopen and I will submit a new pull that fixes these problems.

MartyG-RealSense, 2020-10-28:
I have reopened.  Thank you @JTrantow

MartyG-RealSense, 2020-11-07:
Making a note to keep this case open for a further time period.

MartyG-RealSense, 2020-12-05:
Adding a note to keep this case open for a further time period.

JTrantow, 2020-12-17:
Closed as evidently no interest in correcting bugs in this function. Not just compiler warnings, this function template is buggy and will fail if it's used for non-bool types or at the end of the buffer.

closed 2020-12-17 by JTrantow