realsenseai/librealsense #5739: How to calculate all 3D points at once using the depth image

Uvindu, 2020-01-29:
Hi,
I would like to know how to calculate 3D point data (X,Y,Z distance) at once using depth image given by the realsense camera. 
____python_code_____
`depth_point = pyrealsense2.rs2_deproject_pixel_to_point(depth_intrinsic, pixel coordinates, depth_distance_to pixel)`
depth_point shape is (3,1)
This function needs to run in a double for loop through rows and columns in order to get the 3D points for all depth image pixels.
Is there any function or matrix operation, that I can use to get the 3D point matrix for the whole image without using for loops? (lets say the function should output (3,rows,columns) shape matrix which includes X,Y,Z distances for all pixel values)

Thank you

dorodnic, 2020-01-29:
I believe you are looking for [`pointcloud` class](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pointcloud.html#pyrealsense2.pointcloud)

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

@Uvindu  Did you get it through with dorodnic's reply? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

@Uvindu Could you please update?Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@Uvindu  If no other questions or issues about this, will close this. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport