realsenseai/librealsense #5742: align color/depth pair on saved dataset

Kevin-Delnoije, 2020-01-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 2.31 | 
| Operating System & Version |    Linux Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.14                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  2                          | 
| Language                            |  python                          | 
| Segment			|                           | 

### Issue Description

I would like to align color and depth images saved as .png and loaded into numpy arrays
but i can only find examples of alignment when there is a stream object that gets frames from a live camera like this:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

is it possible to align with a pair of numpy arrays of color and depth?

RealSenseCustomerSupport, 2020-02-08:
----------------------------------------------

Hi @Kevin-Delnoije,

Can you please clarify what is the final objective of this operation? To have a numpy array that represents aligned depth to color image, to convert the result of an alignment to some specific format, etc?

Thank you!

Kevin-Delnoije, 2020-02-10:
yes, the final objective is to align depth to color image, and then estimate distance of detected object.
[This](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) notebook does almost exactly what i want, except that it requires rosbag.
Is it possible to achieve the same with depth and color images only and perhaps a config file about camera settings?

jb455, 2020-02-10:
You would need the intrinsics of both the depth and colour cameras and the extrinsics between them, and the depth scale. 
Note, saving a colourised depth image as png won't allow you to recover the raw depth values for alignment - you would need to save in some format which stores each pixel as a ushort or something similar.

Kevin-Delnoije, 2020-02-10:
thanks that helps a lot,
I am saving the raw numpy array obtained from frame which has data type uint16 and is measured in millimeters for my current realsense camera. 
If you know any examples (preferably python) that use the intrinsics and extrinsics for computing depth-alignment that would be greatly appreciated

jb455, 2020-02-10:
If you're not too worried about latency you can do:

1. [`rs2_deproject_pixel_to_point`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L67) using depth intrinsics
2. [`rs2_transform_point_to_point`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L126) using depth -> colour extrinsics to transform the point to the colour viewpoint
3. [`rs2_project_point_to_pixel`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L19) with the colour intrinsics to get the colour pixel coordinates of the point.

You would need to do this on each pixel you want to align.
I don't know if any of these are available in the python wrapper (they're not in C#) but they're mostly just arithmetic so should be easy to translate.

Kevin-Delnoije, 2020-02-10:
That sounds like a solution! 
For now i am sticking with rosbag files since we are mostly using the color data in the dataset anyway.
ill update this issue if i decide to implement this

closed 2020-02-10 by Kevin-Delnoije