realsenseai/librealsense #5744: "No RealSense Camera Found" error using a demo of labview SDK wrapper library

abdulhafiz7794, 2020-01-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   {Win (10 home) / Version 10.0.18363 Build 18363 | 
| Kernel Version (Linux Only)    |-----| 
| Platform                                  | PC |
| SDK Version                            |  { legacy / 2.32.0 }  /  { legacy / 2.31.0 }  | 
| Language                                |  LabVIEW 2019 32 and 64 bit  | 
| Segment			          |  { LabVIEW 2019 32 and 64 bit}  | 

### Issue Description
"No RealSense Camera Found" error using a demo of labview SDK wrapper library

I'm trying to access intel realsense D435i using LabVIEW wrappers for SDK. 
The camera is recognized by the PC and realsense viewer in USB 3.2 and the streams of all the sensors can be viewed in realsense viewer and depth quality tool.
However, when I run the "Depth and color. vi", " Depth and monochrome.vi " and other demos, I get this same errors as following 
Init err: **No RealSense Camera Found**.
config error: **Error calling rs2_get_stream_profiles (sensor:nullptr):null pointer passed for argument 
                     "sensor"  
                     No Error**
Start err: **No Error
                  No Error
                  Error calling rs2_start_queue (sensor:nullptr, queue:000002160947BBA0):
                  null pointer passed for argument "sensor"** 

What I have tried:
I changed a number of device index and sensor indexes.
I tried both labview 2019 32 bit and 64 bit using both realsense2.dll for 32/64 bit.
I couldn't try labview 2016 because the demo VIs were updated/saved in later versions of LabVIEW.
I tried using different versions of SDK2 such as  l{egacy / 2.32.0 }  /  { legacy / 2.31.0 }.
I tried using all versions of Labview wrappers' library available in GITHUB website:
               (Latest) RealSense SDK2 for LabVIEW 1.0, 2.11.0, May 2018
               RealSense SDK2 for LabVIEW 1.0, 2.10.4, Apr 2018
               RealSense SDK2 for LabVIEW 1.0, 2.8.1, Nov 2017
I also tried updated demos specific for D435 found from intel forum's post
https://forums.intel.com/s/question/0D50P0000490WHVSA2/realsense-d435-and-labview?language=en_US

**please find the attached screenshots of VIs, Device manager, and realsense viewer info.
On the realsense viewer info, I can see the camera is locked. However, I could access the color camera using IMAQ-dx.**
[image]
[image]
[image]
[image]
[image]
[image]

**I tried all those thing but no luck. I would appreciate any help on this issue.**

abdulhafiz7794, 2020-01-29:
[image]

agrunnet, 2020-02-05:
That is indeed odd. If it works in the viewer, it should work in Labview, which is simply a wrapper on realsense2.dll. 

You appear to have done the right thing, but let me outline the exact steps that should work:
1. Run the RS Viewer first and test that it works. 
2. Then find the app's folder location, and more specifically the location of the realsense2.dll file. 
3. Copy that file into the location where the Labview apps reference the realsense2.dll (overwriting the existing one).  This will make sure you are using the latest and identical version of LibRealSense. 

You have to do this last step when the labview VI is closed or you wont be able to overwrite the existing realsense2.dll.

I will look into posting the newest updated package just in case this does not work for you. I have many D435i's and they all work, so we should be able to get to the bottom of this.

agrunnet, 2020-02-08:
[image]
Here is screenshot showing D435i working. We will upload latest version next week. But basically I followed the steps I indicated above.

dorodnic, 2020-02-18:
@abdulhafiz7794 
New version was uploaded, please give it a try

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi @abdulhafiz7794 Do you have a chance to try it out?

Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-03-11 by RealSenseCustomerSupport

YangYunfang, 2023-08-29:
I have the same problem with him.

MartyG-RealSense, 2023-08-29:
Hi @YangYunfang  Can you provide information about the questions below, please?

1.  Which RealSense camera model are you using (for example, D435i)

2.  Which RealSense SDK version are you using?

3.  Are you using the 32-bit or 64-bit DLL?