realsenseai/librealsense #5748: librealsense unsupport Kernel version 5.3.0-28-generic

T2honda, 2020-01-30:
I am trying to install the "librealsense" on Desktop PC with Ubuntu 18.04.3 LTS.

However, I got unsupport kernel error when I run the command " ./scripts/patch-realsense-ubuntu-lts.sh ".

Please see the attachment figure.
[image]

The kernel of the Ubuntu is " 5.3.0-28-generic ", it seems like The Bionic patches are maintained for Ubuntu LTS with kernels 4.15, 4.18 and 5.0 only.

Can anyone please give me some suggestion on how to solve the problem ?

Thank you in advance.

ev-mp, 2020-02-02:
@T2honda hello,
The official support for LTS kernel 5.3 will be added in the next release.
In the meantime you can use the [pending PR](https://github.com/IntelRealSense/librealsense/pull/5751). Feedback will be highly appreciated

christian-rauch, 2020-02-04:
@ev-mp Any hints when this release will happen?

ev-mp, 2020-02-04:
The next release is scheduled for this week. We're planning to have it complete by Monday.

christian-rauch, 2020-02-04:
> The next release is scheduled for this week. We're planning to have it complete by Monday.

Do you mean it is scheduled for next week? Because Monday was yesterday :-)

korzen, 2020-02-05:
Hi, just to confirm that I have successfully patched and built the library under 5.3.0-28-generic using the development branch. My D415 finally works well!

christian-rauch, 2020-02-06:
I received a Debian package update today with the new version and it correctly builds the `librealsense2-dkms` for `5.3.0-28-generic`.

closed 2020-02-06 by RealSenseCustomerSupport

d3-worgan, 2020-02-28:
Hi can anyone advise how to continue in this situation?  Appears to be similar to the one described here i.e. have followed the installed steps described at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md and am now stuck on running step 4, "./scripts/patch-realsense-ubuntu-lts.sh"

[image]

Thanks
Dan