realsenseai/librealsense #5749: XCODE App Bridging C header files

omarfakhr, 2020-01-30:
| Camera Model                       | { D400  | 
| Firmware Version                   |  | 
| Operating System & Version |   MacOS  | 
| Platform                                 | XCODE |
| SDK Version                            | ?                     | 
| Language                            |  C to Swift  | 
| Segment			|  others | 

Hi there,

I managed to run the EXEC files on XCODE. Now I want to make a GUI using Swift and implement the EXEC Target into it. Or any other approach that can work. Here is what I did:

Approach 1:
 I created a bridging header for the C header files in order to have the functions in SWIFT. I am trying to copy the program rs-capture.cpp in SWIFT. I tested the first functions, like:

let colormap = rs2_create_colorizer(nil)

However I get this error: Undefined symbols for architecture x86_64:
  "_rs2_create_colorizer"

Approach 2:

I try to implement the EXEC Target into an App-Target. Do not know how this works, though. Any help?

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

I think you should add librealsense2.dylib or librealsense2.a to your project as a dependence.
Please refer to this document.
https://help.apple.com/xcode/mac/current/#/dev51a648b07

omarfakhr, 2020-02-14:
@RealSenseCustomerSupport Thanks, but where do I find librealsense2.dylib?

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

Please follow this document.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

omarfakhr, 2020-02-15:
@RealSenseCustomerSupport thanks again for the prompt reply. Yes I have tried that. I linked the library into the project and it still does not work. Also the link you suggested is not complete I fear. in order to perform CMAKE the ropistory should be cloned into the Build Folder, which the document does not mention. So I am asking: Is there anyway to have a start to finish complete explanation of how to work with it? I have wasted three weeks or more trying to make this work. So I will appreciate all the help I can get.
Thanks

RealSenseCustomerSupport, 2020-02-16:
----------------------------------------------

according to the issue https://github.com/IntelRealSense/librealsense/issues/4680, I think there is something wrong in your project setting.

omarfakhr, 2020-02-16:
@RealSenseCustomerSupport Thanks again very much. I do not know of this issue addresses my problem. I was able to integrate the library into the XCODE project as you suggested. How can I access the functions? Should I still make a bridging header and access the code from cpp file, while everything else is in swift??

closed 2020-02-27 by omarfakhr