realsenseai/librealsense #5755: UWP 400 series code examples?

brettmilinski, 2020-01-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| Language                            |  C#                         | 

### Issue Description
I just recently downloaded the UWP driver for my cameras, but I cannot get the app to work.

I could not find any code examples of people using the D435 in a UWP app. Are there any available that would show how to capture depth frames and access the data? Is it drastically different than using the SDK?

MartyG-RealSense, 2020-01-31:
May I confirm please that you are using the version of the UWP driver created for the RealSense 400 Series cameras?

https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

brettmilinski, 2020-01-31:
@MartyG-RealSense Yes that is correct, I downloaded the latest version of the driver from Intel.

MartyG-RealSense, 2020-01-31:
Regarding examples, there isn't much for UWP, but the link below may provide at least one usable sample program.

https://github.com/IntelRealSense/librealsense/issues/2722

brettmilinski, 2020-01-31:
@MartyG-RealSense So I have to use a third party API in order to access the cameras and their data in UWP?

MartyG-RealSense, 2020-01-31:
There is little reference information on using UWP with RealSense cameras because it has not been widely used.  The most likely practical application for UWP so far has been to interface a RealSense camera with a Microsoft HoloLens headset.

A couple of the Intel RealSense  team most likely to be able to provide technical advice on this subject are @ev-mp or @dorodnic

closed 2020-02-04 by brettmilinski