realsenseai/librealsense #5759: Unable to capture depth frames using SR305

gitacct1800, 2020-02-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Firmware Version                   | (3.27.3.0) | 
| Operating System & Version |   {Win 10}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.31.0.1234 }                          | 
| Language                            |  { python }                          | 
| Segment			|  { others }                          | 

### Issue Description
<Hi, I'm new to the RealSense camera. I have a problem capturing depth frames from the camera. The error message is shown:

  File "C:/Users/pj/Desktop/handpose/realtimehandpose2.py", line 16, in <module>
    pipeline.start()

RuntimeError: 
Failed to resolve the request: 
	Format: Z16, width: 640, height: 480

Into:
	Formats: 
	 INZI

And below the code snippet:

# First import the library
import pyrealsense2 as rs

#config = rs.config()
#config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Create a context object. This object owns the handles to all connected realsense devices
pipeline = rs.pipeline()
pipeline.start()

try:
    while True:
        # Create a pipeline object. This object configures the streaming camera and owns it's handle
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        if not depth: continue

        # Print a simple text-based representation of the image, by breaking it into 10x20 pixel regions and approximating the coverage of pixels within one meter
        coverage = [0]*64
        for y in xrange(480):
            for x in xrange(640):
                dist = depth.get_distance(x, y)
                if 0 < dist and dist < 1:
                    coverage[x/10] += 1

            if y%20 is 19:
                line = ""
                for c in coverage:
                    line += " .:nhBXWW"[c/25]
                coverage = [0]*64
                print(line)

finally:
    pipeline.stop()

May I know how this can be resolved? Thank you.
>

MartyG-RealSense, 2020-02-02:
It looks as though your script is based on the **Tutorial_1** example from the Python wrapper page, which generates an ASCII art representation of the depth stream using a 640x480 resolution.

https://github.com/leggedrobotics/librealsense-rsl/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

I notice that compared to your script, the example introduces the **try** instruction in a different place.  Could you put it directly above the  **pipeline = rs.pipeline()** line please like in the example script and see if this corrects the problem?

[image]

gitacct1800, 2020-02-06:
I was able to resolve the problem by shifting the try instruction. Thanks for your help.

closed 2020-02-06 by gitacct1800