realsenseai/librealsense #5760: Python: Unable to change camera options after pipeline start T265

RiccardoBe, 2020-02-02:
Hi all,

I have some issues with changing option on my T265.

Whit this code
    pipe = rs.pipeline()

    cfg = rs.config()

    profile = cfg.resolve(pipe)
    sensor = profile.get_device().query_sensors()[0]
    sensor.set_option(rs.option.enable_relocalization, 0)
    sensor.set_option(rs.option.enable_pose_jumping, 0)
    enable_relocalization = sensor.get_option(rs.option.enable_relocalization)
    print("INFO: Relocalization is ", enable_relocalization)

    cfg.enable_stream(rs.stream.pose)  

    pipe.start(cfg)`

I get this error:

```
       pipe.start(cfg)
 RuntimeError: No device connected
```

And if I start the pipeline immediately doing this
`    profile = pipe.start(cfg)
    sensor = profile.get_device().query_sensors()[0]
    sensor.set_option(rs.option.enable_relocalization, 0)
    sensor.set_option(rs.option.enable_pose_jumping, 0)
    enable_relocalization = sensor.get_option(rs.option.enable_relocalization)
    print("INFO: Relocalization is ", enable_relocalization)

    # Enable the stream we are interested in
    cfg.enable_stream(rs.stream.pose) `

I get:

```
     sensor.set_option(rs.option.enable_relocalization, 0)
RuntimeError: Option is read-only while streaming
```

I am using latest librealsense, any ideas? Thanks

neilyoung, 2020-02-02:
For the first issue see here https://github.com/IntelRealSense/librealsense/issues/5479#issuecomment-567448946 and the second is already explained in the error message: You can't change options, while streaming.

Proper sequence is:

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.pose)
device = config.resolve(pipeline).get_device()
pose_sensor = device.first_pose_sensor()
# Do everything with options here
pose_sensor.set_option(rs.option.enable_map_preservation, 1)
pipeline.start(config)
```

RiccardoBe, 2020-02-02:
@neilyoung Thanks for the quick answer! It works, the second example is self explanatory as you said but I tried (almost) everything to try to make it work :) 
The issue can be closed.

closed 2020-02-02 by RiccardoBe