realsenseai/librealsense #5761: Align recorder (RGB & D ) SR305 VS D435i

devshank3, 2020-02-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300 / SR305 } | 
| Firmware Version                   | 3.27.3.0| 
| Operating System & Version |   Win (10) | 
| Kernel Version (Linux Only)    |  -                                        | 
| Platform                                 | PC |
| SDK Version                            | 2.31.0.1234 }                          | 
| Language                            |  python                   | 
| Segment			|  3D - scanning                        | 

### Issue Description
Hello Team Intel Realsense ,

I'm new to realsense and 3D reconstruction , my query is regarding the depth frame captured between D435i and SR305 depth cameras 

Im using Open3D's reconstruction pipeline  

1. https://github.com/intel-isl/Open3D/tree/master/examples/Python/ReconstructionSystem
2. http://www.open3d.org/docs/release/tutorial/ReconstructionSystem/index.html

and for custom data using Intel realsense , Im using Open3D's capture code 

1.http://www.open3d.org/docs/release/tutorial/ReconstructionSystem/capture_your_own_dataset.html#capture-your-own-dataset
2. https://github.com/intel-isl/Open3D/blob/master/examples/Python/ReconstructionSystem/sensors/realsense_recorder.py

with data captures using D435i it was giving awesome High quality 3D results 
But with SR305 using same code the 3D mesh result was not okay (no results)

My question :  Is there some thing with the format 
because in D435i  depth data the near objects are darker and in SR305 depth data it was greyish 

### D435i sample (640 * 480)
### Color
[image]

### Depth
[image]

### SR305 sample (640 * 480)
### Color
[image]
### Depth
[image]

Same real sense capture code was used 
https://github.com/devshank3/Intel-realsense-issues/blob/master/SR305/realsense_recorder.py

Please let me know how this can be resolved and how may i contribute 
Thank You

RealSenseCustomerSupport, 2020-05-02:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

devshank3, 2020-05-03:
Yes still I have this issue....

The reconstruction was not proper , Is there any proper way to get RGB and its aligned depth ?
I used the above mentioned real sense recorder code .

liushuya7, 2020-12-12:
I have the same issue. I want to use SR305 to perform closer scans for 3D reconstruction, however, the depth images seem not right compared to D435.

MartyG-RealSense, 2021-09-11:
Hi @devshank3  I apologize that you did not receive a further response from us.  Do you still require assistance with this case, please?  Thanks!

devshank3, 2021-09-11:
Nope I'm long way out of that project ....
Thanks @MartyG-RealSense

closed 2021-09-11 by devshank3