realsenseai/librealsense #5762: Installation proceeded according to the installation procedure of librealsense2. When I run sudo apt-get install librealsense2-utils I'm having a problem with the following error: The following packages have unmet dependencies. Does anyone know the cause or solution?

UMINODAIKI, 2020-02-02:
Installation proceeded according to the installation procedure of librealsense2. 
When I run 

sudo apt-get install librealsense2-utils

I'm having a problem with the following error: 

The following packages have unmet dependencies.

Does anyone know the cause or solution?

camera model ： D435i T265i
Firmware Ver: As of 2020.2.2 Latest
OS Ubuntu: Ubuntu 16.04
SDK Ver II: Intel® RealSense ™ SDK 2.0 (build 2.32.1)
Referenced page: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
PC: GALLERIA GCR1660TGF

The final goal is to run 

roslaunch realsense2_camera rs_camera.launch

Running realsense on ROS.
In the past, 

roslaunch realsense2_camera rs_camera.launch and other nodes related to realsense were successfully executed.
As you work, you'll see errors when you run the node.
So I tried to uninstall and reinstall librealsense2.

Here are the steps I performed:

(1) Uninstall librealsense2 with the following command

   dpkg -l | grep "realsense" | cut -d "" -f 3 | xargs sudo dpkg --purge
   sudo apt autoremove librealsense2 *

   https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

(2) Reinstall using the following URL

   https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
   
(3) When I run sudo apt-get install librealsense2-utils, the following error was displayed

$ sudo apt-get install librealsense2-utils
[sudo] password for nvidia:
Reading package lists ... Done
Building dependency tree
Reading state information ... Done
Some packages could not be installed.This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils: Depends: librealsense2 (= 2.32.1-0 ~ realsense0.1913) but it is not going to be installed
                       Depends: librealsense2-gl (= 2.32.1-0 ~ realsense0.1913) but it is not going to be installed
                       Depends: libc6 (> = 2.27) but 2.23-0ubuntu11 is to be installed
                       Depends: libstdc ++ 6 (> = 6) but 5.4.0-6ubuntu1 ~ 16.04.12 is to be installed
E: Unable to correct problems, you have held broken packages.

(4) I thought that the first installation was in the way.
The directories and files containing the strings of librealsense2, librealsense2-gl, libc6, and libstdc ++ 6 were manually deleted.

Ex.sudo apt autoremove libc6 *

(5) When I run sudo apt-get install librealsense2-utils, I got the error of (3) again

(6) Ignored the above error and proceeded with the installation procedure. I tried to run realsense-viewer 、The following error was displayed.

$ realsense-viewer
realsense-viewer: error while loading shared libraries: librealsense2-gl.so.2.32: cannot open shared object file: No such file or directory

The final goal is to run 

roslaunch realsense2_camera rs_camera.launch

Running realsense on ROS.
Does anyone know the cause or solution?
I do not have much knowledge. Thank you for your cooperation.

ev-mp, 2020-02-03:
@UMINODAIKI hello,
You need to correct the repo source files as currently you're trying to download and install Librealsense debians on Ubuntu16 using the repository for Ubuntu18 (build number 1913 is for Bionic).
>The following packages have unmet dependencies:
librealsense2-utils: Depends: librealsense2 (= 2.32.1-0 ~ realsense0.**1913**) but it is not going to be installed

Follow the installation guide - https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages to register the proper repository for Ubuntu16 and remove the registration for Ubuntu 18 to resolve the inconsistency

ev-mp, 2020-02-18:
@UMINODAIKI , do you need further assistance with this issue? 
Please update.

UMINODAIKI, 2020-02-19:
@ev-mp  Sorry for the late reply.After removing the repository for Bionic, I was able to reinstall it successfully. Thank you in the future.

ev-mp, 2020-02-19:
Thank you for the feedback!

closed 2020-02-19 by ev-mp