realsenseai/librealsense #5763: Capturing a color image after aligning frames to the depth stream

grigala, 2020-02-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Language                            | Python                           | 
| Segment			|  others                           | 

Hello everyone, 

If I have frames aligned to depth stream the color image captured after this alignment contains black spots(because of missing depth data). Is there a way to obtain a normal color image in this setting without having those black marks and without losing alignment to the depth stream? 

Thanks in advance.

MartyG-RealSense, 2020-02-02:
Do the black spots look similar to the ones in the  discussion linked to below, please?

https://github.com/IntelRealSense/librealsense/issues/2445

grigala, 2020-02-03:
Hi MartyG, 

No, it's more like this: https://github.com/IntelRealSense/librealsense/issues/3045#issuecomment-453624096
[image]

This is explainable - because frames are aligned to depth, the color image also inherits non-existing depth pixels, hence black spots. But my question is can I get a normal color image this way and don't lose alignment? I can always capture a good color image from non-aligned color stream but then my color image and point-cloud will not be aligned.

Or is there a way to manually align captured color frame to the depth frame given extrinsics? Like applying an extrinsic(don't know which extrinsics) parameter matrix maybe?

MartyG-RealSense, 2020-02-03:
When you said black spots, I thought you literally meant spots (dots).  Are you referring to a black 'area' please?

If so, the outline around the hand looks like an effect called "occlusion", where the depth and color cameras are in different physical positions and so can't see the all the same parts of the scene.

A user who was apparently experiencing occlusion in their image posted the code they used to solve their particular problem.

https://github.com/IntelRealSense/librealsense/issues/2092#issuecomment-406332447

jb455, 2020-02-03:
Is there a reason you can't align depth to colour and use the normal colour image? Alternatively, you can get a colour image (of sorts) naturally aligned to depth if you query the IR frame (only D415, not 435).

grigala, 2020-02-03:
Hi @MartyG-RealSense, sorry yes, I meant a black area(s) displayed on a color image. 

The link you posted aligns frames to the color stream which works just fine for me(/cc @jb455) when it comes to aligning frames. However @jb455, after this process, if I calculate points from the depth frame it does not produce a point cloud that is aligned to color image.  I tested it by taking a 2D landmark from a color image(i.e. tip of the nose) and projecting it to a 3D point using `rs2_deproject_pixel_to_point` - the projected point is a bit off. 
```python
align_to = rs.stream.color # <- aligned to color stream
align = rs.align(align_to)
    try:
        while True:
            frames = pipeline.wait_for_frames()
            aligned_frames = align.process(frames)
            depth_frame = aligned_frames.get_depth_frame()
            color_frame = aligned_frames.get_color_frame()

            # Those points are not aligned to color frame idk why
            pts = pc.calculate(depth_frame) 
```
Therefore to avoid this misalignment I am aligning frames to depth stream, in that way the previous test passes. 
```python
align_to = rs.stream.depth # <- aligned to depth stream
align = rs.align(align_to)
    try:
        while True:
            frames = pipeline.wait_for_frames()
            aligned_frames = align.process(frames)
            depth_frame = aligned_frames.get_depth_frame()
            color_frame = aligned_frames.get_color_frame()

            pts = pc.calculate(depth_frame)
```
Which leads us to the problem I described initially if I capture color image :
```python
 color_image = np.asanyarray(color_frame.get_data())
```
it will contain black areas and it is not just because of occlusion phenomena but generally, pixels whose depth data cannot be obtained are blacked out. I am wondering whether hole filling filter would help or not but I haven't experimented with filters yet.

jb455, 2020-02-03:
What's your test which suggests the point cloud isn't aligned? Are you using the correct intrinsics in `rs2_deproject_pixel_to_point`? You should pass the colour intrinsics. 
There was a bug a long time ago where the point cloud module was using the wrong intrinsics on aligned frames but if you're using a fairly recent version that shouldn't be a problem.

If you want to stick with aligning to depth, hole filling may help, but generally it won't be able to align pixels without depth.

grigala, 2020-02-03:
Thanks @jb455 , this might be a problem, I was passing depth intrinsics.

> Are you using the correct intrinsics in `rs2_deproject_pixel_to_point`? You should pass the colour intrinsics.

closed 2020-02-03 by grigala