realsenseai/librealsense #5764: how to synchronize the Depth and color frames ?

nawara72, 2020-02-02:
Hi 
I would appreciate any help on what I am doing wrong. I am trying to get the timestamps of both depth and color frame of a D435, but they appear not synchronized. Snippet of my code and outputs are shown below. Would appreciate any help/comments. Thanks

depth_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
depth_sensor.set_option(RS2_OPTION_EXPOSURE, 8500);
depth_sensor.set_option(RS2_OPTION_GAIN, 16);
depth_sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE, 1);

color_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
color_sensor.set_option(RS2_OPTION_EXPOSURE, 100); 
color_sensor.set_option(RS2_OPTION_GAIN, 64);
color_sensor.set_option(RS2_OPTION_FRAMES_QUEUE_SIZE, 1);

rs2::pipeline pipe;
rs2::config cfg;
cfg.enable_device(serial);
cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_RGB8, 30);

rs2::pipeline_profile profile = pipe.start(cfg);
for (int frame_count = 0; frame_count < 500; ++frame_count) {
	rs2::frameset frames = pipe.wait_for_frames();
	rs2::frame color_frame = frames.get_color_frame();
	rs2::frame depth_frame = frames.get_depth_frame();
	std::cout << frames.size() << " frames from " << serial << ": ";
	std::cout.precision(std::numeric_limits<double>::max_digits10);

std::cout << “Depth : " << std::scientific << depth_frame.get_timestamp()	<< "(" << depth_frame.get_frame_number() << "), ";
	std::cout << "Color : " << std::scientific << color_frame.get_timestamp() << "(" << color_frame.get_frame_number() << "), ";
}

Outputs:
DEPTH: 1.58068341329593359e+12(22)
COLOR : 1.58068341322219092e+12(11)
DEPTH: 1.58068341332935083e+12(23)
COLOR : 1.58068341325554907e+12(12
DEPTH: 1.58068341336267529e+12(24)
COLOR : 1.58068341328890576e+12(13)
DEPTH: 1.58068341339593872e+12(25)
COLOR : 1.58068341332226245e+12(14
DEPTH: 1.58068341342929370e+12(26)
COLOR : 1.58068341335562012e+12(15)
DEPTH: 1.58068341346285474e+12(27)
COLOR : 1.58068341338897974e+12(16)
DEPTH: 1.58068341349597192e+12(28)
COLOR : 1.58068341342233862e+12(17)
DEPTH: 1.58068341352929688e+12(29)
COLOR : 1.58068341345569849e+12(18)
DEPTH: 1.58068341356293750e+12(30)
COLOR : 1.58068341348905615e+12(19)
DEPTH: 1.58068341359607446e+12(31)
COLOR : 1.58068341352241382e+12(20)
DEPTH: 1.58068341362944995e+12(32)
COLOR : 1.58068341355577124e+12(21)
DEPTH: 1.58068341366298633e+12(33)
COLOR : 1.58068341358912842e+12(22)

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi, please check the TimestampDomain (hardware clock, system time or global time).

nawara72, 2020-02-09:
It says system time, would appreciate any help on what to do, Thanks

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

The global timestamp you provided seems have much improved the delta between depth and color frame:

Before:

    System Time
    DEPTH: 1580683413**562**.93750 ms
    COLOR: 1580683413**489**.05615 ms

    DEPTH: 1580683413596.07446 ms
    COLOR: 1580683413522.41382 ms

    DEPTH: 1580683413629.44995 ms
    COLOR: 1580683413555.77124 ms

    DEPTH: 1580683413662.98633 ms
    COLOR: 1580683413589.12842 ms

After:

    Global Time
    DEPTH: 158127832023**3**.93579 ms
    COLOR: 158127832023**8**.59668 ms

    DEPTH: 1581278320267.28247 ms
    COLOR: 1581278320271.94824 ms

    DEPTH: 1581278320300.62891 ms
    COLOR: 1581278320305.29956 ms

    DEPTH: 1581278320333.97510 ms
    COLOR: 1581278320338.63989 ms

nawara72, 2020-02-12:
Thanks for your help, yes you are right delta has improved but the frame counter is off. Should I be using the frame counters for any sort of synchronization indications??

Thanks

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

The frame counter is due to the rgb and depth sensors started at slightly different time.
Closing this issue if no further question. Thank you.

nawara72, 2020-02-13:
Great thanks for all the help

closed 2020-02-13 by nawara72