realsenseai/librealsense #5765: Build librealsense with BUILD_PYTHON_BINDINGS ?

jiapei100, 2020-02-03:
```console
 CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
   Python config failure:

   Fatal Python error: Py_Initialize: Unable to get the locale encoding

   ModuleNotFoundError: No module named 'encodings'

   Current thread 0x00007f13e8b66340 (most recent call first):

 Call Stack (most recent call first):
   wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
   wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)
```

I installed my own Python and pybind11:

```console
➜  ~ python --version
Python 3.8.1
➜  ~ pip show pybind11
Name: pybind11
Version: 2.4.3
Summary: Seamless operability between C++11 and Python
Home-page: https://github.com/pybind/pybind11
Author: Wenzel Jakob
Author-email: wenzel.jakob@epfl.ch
License: BSD
Location: ~/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
➜  ~ 

```

How can I install librealsense with my own pybind11 and with BUILD_PYTHON_BINDINGS enabled?

Cheers
Pei

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@jiapei100 It might be the similar issue as that in the link below. Please refer to the solution to see it helps.

https://stackoverflow.com/a/49977872

jiapei100, 2020-02-11:
@RealSenseCustomerSupport 

- If I `export PYTHONHOME=/usr/bin`, I got the following **ERRORs**.

```console
➜  build git:(master) ✗ echo $PYTHONHOME
/usr/bin
➜  build git:(master) ✗ echo $PYTHONPATH
/home/longervision/.local/lib/python3.8/site-packages:/usr/lib/python3.8/site-packages:/usr/local/lib/python3.8/site-packages
➜  build git:(master) ✗ pip show setuptools
Python path configuration:
  PYTHONHOME = '/usr/bin'
  PYTHONPATH = '/home/longervision/.local/lib/python3.8/site-packages:/usr/lib/python3.8/site-packages:/usr/local/lib/python3.8/site-packages'
  program name = '/usr/bin/python'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/usr/bin/python'
  sys.base_prefix = '/usr/bin'
  sys.base_exec_prefix = '/usr/bin'
  sys.executable = '/usr/bin/python'
  sys.prefix = '/usr/bin'
  sys.exec_prefix = '/usr/bin'
  sys.path = [
    '/home/longervision/.local/lib/python3.8/site-packages',
    '/usr/lib/python3.8/site-packages',
    '/usr/local/lib/python3.8/site-packages',
    '/usr/bin/lib/python38.zip',
    '/usr/bin/lib/python3.8',
    '/usr/bin/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00007f12b17cbd80 (most recent call first):
<no Python frame>
```

- If I `export PYTHONHOME="" `

```console
➜  build git:(master) ✗ export PYTHONHOME=""
➜  build git:(master) ✗ pip show setuptools
Name: setuptools
Version: 45.2.0
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
Location: ~/.local/lib/python3.8/site-packages
Requires: 
Required-by: zope.interface, wadllib, twine, tensorboard, Sphinx, protobuf, Markdown, lazr.restfulclient, launchpadlib, kiwisolver, jsonschema, ipython, google-auth
➜  build git:(master) ✗ 
```

That seems to be a **Python** bug....

jiapei100, 2020-02-11:
@RealSenseCustomerSupport 

Ok, I got one answer from [https://discuss.python.org/t/fatal-python-error-init-fs-encoding-failed-to-get-the-python-codec-of-the-filesystem-encoding](https://discuss.python.org/t/fatal-python-error-init-fs-encoding-failed-to-get-the-python-codec-of-the-filesystem-encoding/3173).

It looks like it's NOT a python bug. Then, it clearly is a bug from librealsense itself. My installed **pybind11** is:

```console
➜  build git:(master) ✗ pip show pybind11
Name: pybind11
Version: 2.4.3
Summary: Seamless operability between C++11 and Python
Home-page: https://github.com/pybind/pybind11
Author: Wenzel Jakob
Author-email: wenzel.jakob@epfl.ch
License: BSD
Location: /home/longervision/.local/lib/python3.8/site-packages
Requires: 
Required-by: 
```

Does it have something to do with the librealsense's own 3rd-party pybind11's **FindPythonLibsNew.cmake** ?

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

It's not a issue.
You have wrong PYTHONHOME value.
https://docs.python.org/3/using/cmdline.html#envvar-PYTHONHOME

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

Will close this issue if no more question. Thanks.

closed 2020-03-20 by RealSenseCustomerSupport