realsenseai/librealsense #5766: Regarding rs2_project_color_pixel_to_depth_pixel

lokeshgs87, 2020-02-03:
Hi @kafan1986

The following is the URL which is written in python: https://github.com/IntelRealSense/librealsense/issues/5603  

Can you please help us to get the following details in android

depth_intrin
color_intrin
depth_to_color_extrin
color_to_depth_extrin

and also following functions (to convert color pixel to depth )

rs2_deproject_pixel_to_point
rs2_transform_point_to_point
rs2_project_point_to_pixel

kafan1986, 2020-02-03:
> Hi @kafan1986
>
> The following is the URL which is written in python: #5603
>
> Can you please help us to get the following details in android
>
> depth_intrin
> color_intrin
> depth_to_color_extrin [...]

These native APIs are not exposed with the current Android librealsense SDK. You need to write your own wrapper in native code and call that via JNI.

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@lokeshgs87 Have submitted your requirement to our engineering team. Any update will let you know. Thanks!

RealSenseSupport, 2020-08-21:
@lokeshgs87 Android intrisic and extrinsic API now added to the Android wrapper in PR#https://github.com/IntelRealSense/librealsense/pull/6594  

depth_intrin
color_intrin
depth_to_color_extrin
color_to_depth_extrin

Below API was also added by PR# https://github.com/IntelRealSense/librealsense/pull/6818

rs2_deproject_pixel_to_point
rs2_transform_point_to_point
rs2_project_point_to_pixe

Please have a try and feedback to us. Thanks!

RealSenseSupport, 2020-09-04:
@lokeshgs87 PR https://github.com/IntelRealSense/librealsense/pull/6594 and PR https://github.com/IntelRealSense/librealsense/pull/6818/ now merged into the latest SDK v2.38.1. Looking forward to your test update. Thanks!

RealSenseSupport, 2020-09-14:
@lokeshgs87 Did you get chance to try SDK v2.38.1? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-21:
@lokeshgs87 Sorry that we didn't get response from you for weeks. As the feature already implemented, we'll close this at this point. Any issue or question please feel free to create another new ticket. Thanks!

closed 2020-09-21 by RealSenseSupport

Tikitaka02, 2023-03-09:
Hello, I would like to know how to call the function in Android studio. Is it equivalent to rs2_project_color_pixel_to_depth_pixel?