realsenseai/librealsense #5767: Depth frames not refreshing

Pallav56, 2020-02-03:
Hi,

While using **box_dimensioner_multicam__demo.py** to calculate the dimension of box, depth frame is not getting refreshed/ changed but color frame is. Please guide me how i can sync both color and depth frame.

I am using **realsense_device_manager** for reading the frames having single camera

Attached the color frame with corresponding depth frame.

[image]
[image]
[image]
[image]

Pallav56, 2020-02-03:
I am using D435i.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Please kindly refer to #3325 and see if the provided fix works in your case.
Thank you.

Pallav56, 2020-02-09:
I am already using the latest code, also able to do the calibration with chessboard, and not getting any error while running on objects, only the depth frame is not in sync in color frame.

Hence this thread [#3325](https://github.com/IntelRealSense/librealsense/issues/3325) does not help.

Pallav56, 2020-02-10:
While capturing the live stream, color frame gets refreshed according to FPS given, but depth frame having the same FPS is not getting refreshed with the same rate as color frame is.

Pallav56, 2020-02-19:
Hi, Kindly Assist as it is an urgent issue, please let me know if you requires more issue.

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

I am not sure why the depth image is not updating. It works for my D435i with firmware version 05.12.02.100.
Is the "align-depth2color.py" or "python-tutorial-1-depth.py" able to stream depth continuously?
Thank you.

Pallav56, 2020-02-20:
Thanks for the update, it is working with "align-depth2color.py" but not with "box_dimensioner_multicam__demo.py" for calculating the box dimensions.

Do we have any other alternate way to calculate the box/ Object dimensions.

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

As a reference, there is a C++ measure example that can be referred to to measure the distance between two points.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

Please also take a look at these developers/software providers that provide 3D scanning application paired with RealSense Depth Cameras.
https://www.intelrealsense.com/software-for-intel-realsense/

Thank you.

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Please let us know if there is any further question on this.
Thank you.

closed 2020-03-06 by RealSenseCustomerSupport