realsenseai/librealsense #5773: Problem of downgrading the librealsense version from 2.32.1 to 2.31.0

drfg0119, 2020-02-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR300 / D400 } | 
| Firmware Version                   |   | 
| Operating System & Version |   Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    | 4.9                                      | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  {  2.32.1 }                          | 
| Language                            |  {C/C# }                          | 
| Segment			|  Robot                          | 

### Issue Description
Hello, I have a problem about downgrading the version of RealSense SDK.
Yesterday I updated the SDK from 2.31.0 to 2.32.1 accidentally, I tried a lot of methods to make it from 2.32.1 to 2.31.0, but I failed ....
Is there any people experienced this ..? Please help me.

and this is the reference I followed: 
[https://github.com/jetsonhacks/installRealSenseSDK](url)

Please excuse me if you may not understand or you feel bad with my poor English QAQ

ev-mp, 2020-02-03:
@drfg0119 hello,
In case you build the SDK from source then you need to first uninstall the v2.32.1 with `sudo make uninstall` invoked from the Librealsense build directory, probably `<path_to_lrs_src>/build/`.

If you were using the Debian packages then [follow the installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages) to remove the current version, then re-install Librealsense SDK using an explicit version/build:
- List all the available versions with `apt-cache showpkg librealsense2`
- Select and install the specific version from the list. For instance, to install the demos run:
    `sudo apt install librealsense2=2.31.0-0~realsense0.tbd librealsense2-gl=2.31.0-0~realsense0.tbd librealsense2-utils=2.31.0-0~realsense0.tbd`
*Make sure to replace `tbd` with the actual build number.

drfg0119, 2020-02-04:
> @drfg0119 hello,
> In case you build the SDK from source then you need to first uninstall the v2.32.1 with `sudo make uninstall` invoked from the Librealsense build directory, probably [...]

drfg0119, 2020-02-04:
> @drfg0119 hello,
> In case you build the SDK from source then you need to first uninstall the v2.32.1 with `sudo make uninstall` invoked from the Librealsense build directory, probably [...]

Hi @ev-mp , I follow your suggestion to make it downgrade successfully !! I'm really appreciate, thank you. And this is my commands after deleting old version:
$ `sudo apt install librealsense2=2.31.0-5ubuntu5~1.gbp95d195 librealsense2-dbg=2.31.0-5ubuntu5~1.gbp95d195 librealsense2-utils=2.31.0-5ubuntu5~1.gbp95d195 librealsense2-gl=2.31.0-5ubuntu5~1.gbp95d195 librealsense2-dev=2.31.0-5ubuntu5~1.gbp95d195`

$ `sudo apt install librealsense2-udev-rules=2.31.0-5ubuntu5~1.gbp95d195`

Btw, I got another problem about the version matched issue(Intel ros wrapper & librealsense), so I deleted the workspace and create a new one after backup all the files (except CMakeList.txt) which are in `/${workspace}/src` , and then I put the all of old files into the new `/${workspace}/src` , run again `catkin_make` , and then, mission completed !!

My equipment info now: 
TX2 version: Jetpack 4.3 Ubuntu 18.04
librealsense version: v2.31.0

ev-mp, 2020-02-04:
@drfg0119 hello, I'm glad that the info was useful.

After installing a new SDK version I believe you must rebuild the ros wrapper from scratch. Additionally a critical step is missing in the description provided:
>Btw, I got another problem about the version matched issue(Intel ros wrapper & librealsense), so **I deleted the workspace** and create a new one..

Did you uninstall the previous ros wrapper version **before** deleting the work space? If not - it is possible that you have a mix of both the old and new binaries, libs and includes existing side by side which you need to solve.
In case you'll need a further support for ROS-related issues you can open an ticket on the [dedicated forum](https://github.com/IntelRealSense/realsense-ros)

drfg0119, 2020-02-04:
> @drfg0119 hello, I'm glad that the info was useful.
>
> After installing a new SDK version I believe you must rebuild the ros wrapper from scratch. Additionally a critical step is missing in the [...]

@ev-mp Hi, I didn't uninstall the previous ros wrapper version, I just delete the work space and rebuild the work space, and it still works (EX: rs_camera.launch , opensourcetracking.launch). Maybe I'll try this method next time when I meet the question. Thank you for your attention!

RealSenseCustomerSupport, 2020-02-08:
----------------------------------------------

Hi @drfg0119,

Do you need any additional help with the original question regarding downgrading the librealsense version from 2.32.1 to 2.31.0?

Thank you!

drfg0119, 2020-02-11:
Hi @RealSenseCustomerSupport , I got a problem about performance of SR305, when I use TX2 to connect it on the API(realsense-viewr_2.31.0), I didn't see the same results on PC. The work distance isn't that good on TX2.
Btw, the SDK version on TX2 I used is the same on PC.

drfg0119, 2020-02-11:
Hi @ev-mp , I got the problem you said, ha ha. I rebuild the Intel ros wrapper and solved the problem finally, thank you!

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

@drfg0119,

Please open a new thread with the question about performance of SR305. The current thread will be closed since the original question regarding downgrading the librealsense version from 2.32.1 to 2.31.0 was resolved.

Thank you!

drfg0119, 2020-02-17:
> @drfg0119,
> 
> Please open a new thread with the question about performance of SR305. The current thread will be closed since the original question regarding downgrading the librealsense version from 2.32.1 to 2.31.0 was resolved.
> 
> Thank you!

OK! Thank you!!

closed 2020-02-17 by drfg0119