realsenseai/librealsense #5774: UWP App crashes when trying to access cameras (D435 and T265).

brettmilinski, 2020-02-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  and T265| 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC|
| SDK Version                            |  2.29.0                           | 
| Language                            |  C#                         | 

### Issue Description
I am trying to access my D435 and T265 cameras in a UWP. The following block of code produces an error:
```C#
using (var ctx = new Context())
            {
                var devices = ctx.QueryDevices();
                foreach (var device in devices)
                {
                    string sn = device.Info[CameraInfo.SerialNumber];
                    string name = device.Info[CameraInfo.Name];
                    //keep track of name
                    names.Add(name);

                    var cfg = new Config();
                    //enable the device
                    cfg.EnableDevice(sn);
                    var pipe = new Pipeline(ctx);
                    pipe.Start(cfg);
                    pipes.Add(pipe);
                }
            }
```
I get the following error:

> System.TypeInitializationException: 'The type initializer for 'Intel.RealSense.Context' threw an exception.'
Inner Exception
PlatformNotSupportedException: Custom marshalers are not supported

The error occurs on the line:
```C#
using (var ctx = new Context())
```
Am I misusing the cameras in UWP?

MartyG-RealSense, 2020-02-04:
Advice given by a RealSense team member on this forum is that there is not UWP support for T265. 

https://github.com/IntelRealSense/librealsense/issues/3758

The RealSense UWP driver supports the 400 Series cameras.

brettmilinski, 2020-02-04:
@MartyG-RealSense I now remember reading this on the forums, but I still wish to use the 400 series in my UWP. Either way the program is crashing before I even try to access the cameras, so I must be trying to instantiate them incorrectly.

MartyG-RealSense, 2020-02-04:
Are you using the RealSense UWP driver please?

https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

brettmilinski, 2020-02-04:
@MartyG-RealSense Yes I have the latest download!

MartyG-RealSense, 2020-02-04:
It's a challenging problem.  There is not much reference material for scripting for UWP and C# other than the link below:

https://github.com/IntelRealSense/librealsense/issues/2702#issuecomment-444035850

@dorodnic from the RealSense team is probably going to be one of the best people to tackle this question, as they have already investigated the subject in the above link.

brettmilinski, 2020-02-04:
@MartyG-RealSense No worries I am going to move forward with another solution.

closed 2020-02-04 by brettmilinski