realsenseai/librealsense #5775: Feature Request: Read the state (or timestamp) of the Sync pin

HowyZax, 2020-02-03:
was: Syncing camera to Velodyne Lidar's Pulse Per Second signal

I would like to resolve time differences between camera frames and external Lidar frames. I could do this if there were a way to access (via the camera's metadata) the state of the SYNC hardware input pin.

By using a one-shot to convert an external PPS (Pulse Per Second) signal (a common output on GPS modules for 1us syncing) to a 1 Hertz square wave, we could create a unified timestamp that allows us to keep track of the timestamp of any frame from an external device. This method would have about 20ms of jitter but frame times could be better resolved over a period of several seconds by dynamically stretching the PPS pulse width to find frame edges.

What would be better would be to have a frame timestamp metadata entry that was labeled "microseconds since last (falling or rising) edge of Sync pin)".  This would allow us to create a microsecond level accurate timestamp of all frames from various devices regardless of whether or not they are hardware-frame-synced together.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Hi, HowyZax,

Currently, we do have frame metadata that store the system or hardware (D5 ASIC) time stamp
but it not meet your need.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf

To synchronize with other camera, we have the pin 5 (sync pin) to other slave devices
the Sync pin is frame based trigger

or all camera can use external trigger (which also frame based trigger, one trigger one frame)

Above is the approached we provided for synchronizations between different camera/devices

any concern/limit that above approaches not adequate  for your application?

in your case, looks you plan to use 1 HZ sync clock to all cameras (Slave) then reset the time stamp of all devices  at either the rising or falling edge for each cycle.  correct?

closed 2020-02-24 by RealSenseCustomerSupport

YoshuaNava, 2021-01-28:
@RealSenseCustomerSupport I would also be interested in this feature. Can this issue be re-opened?