realsenseai/librealsense #5776: error: for statement expected before ‘uint16_t’

jiapei100, 2020-02-04:
I'm quite sure `#pragma omp parallel for schedule(dynamic)` brings me the following **ERROR** message if I build with **kinfu** enabled, for if I comment out this line 263 in file **rs-kinfu.cpp**, it compiles successfully.

```console
....../librealsense/wrappers/opencv/kinfu/rs-kinfu.cpp: In lambda function:
....../librealsense/wrappers/opencv/kinfu/rs-kinfu.cpp:266:17: error: for statement expected before ‘uint16_t’
                 uint16_t* p_depth_frame = reinterpret_cast<uint16_t*>(const_cast<void*>(d.get_data()));
```

How to use OpenMP for kinfu in librealsense?

cheers
Pei

eokeeffe, 2020-06-16:
Two things, I was able to fix the issue with the compilation, just swap the lines as such, then add
the depth_frame as a shared resource

uint16_t* p_depth_frame = reinterpret_cast<uint16_t*>(const_cast<void*>(d.get_data()));
#pragma omp parallel for shared(p_depth_frame) schedule(dynamic)

secondly you'll want to make sure you have this in the console or your .bashrc file if using opencv 4.x and above
export OPENCV_OPENCL_DEVICE_MAX_WORK_GROUP_SIZE=512

jiapei100, 2020-08-29:
@eokeeffe 

Thank you very much ... You saved my life... Thanks...

closed 2020-08-29 by jiapei100