realsenseai/librealsense #5777: pyrealsense2 Python3.8.1 support?

jiapei100, 2020-02-04:
Check this
**https://pypi.org/project/pyrealsense2/#files**, wheel for python3.8.1 is NOT listed yet.
Is that IntelRealSense who has been maintaining **https://pypi.org/project/pyrealsense2/#files**?

Cheers
Pei

dorodnic, 2020-02-05:
At the moment we provide pypi packages for Python 2, 3.6 and 3.7. 
You have the option of switching to one of the supported python versions or building the library from source with python 3.8

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

@jiapei100 With the reply from dorodnic, any other questions? Looking forward to your reply. Thanks!

jiapei100, 2020-02-13:
@dorodnic @RealSenseCustomerSupport 
If I insist on using Python 3.8.1? Or, I wanna build librealsense from scratch with Python bindings Enabled?

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@jiapei100 If you want build librealsense from source, please refer to https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@jiapei100 Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@jiapei100 Sorry we didn't get response from your side for weeks. Will close the issue at this point. Please feel free to create new one if you still have issues or questions. Thanks!

closed 2020-03-04 by RealSenseCustomerSupport