realsenseai/librealsense #5779: inaccurate distance when using API get_distance

ShettyHarapanahalli, 2020-02-04:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|844212071706|
|**Firmware Version**|05.12.02.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|904523050880|
|**Firmware Update Id**|904523050880|

Hi, 
I have been trying to get the distance of an object at a particular pixel, I get inaccurate distance
as compared with the same pixel using the realsense-viewer tool. I have attached the code that I have been using below - 

`import pyrealsense2 as rs
import numpy as np
import cv2

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 30)

depth_frame = None
color_image = None

def mouse_drawing(event, x, y, flags, params):
    try:
        global pos
        if event == cv2.EVENT_LBUTTONDOWN:
            dist = depth_frame.get_distance(int(x), int(y))
            print(f"\r Distance at the selected point x:{x},y:{y} is: {int(dist*1000)} mm")
    except Exception as e:
        print(e)

# Start streaming
pipeline.start(config)
cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
cv2.setMouseCallback("RealSense", mouse_drawing)

try:

    while True:
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', color_image)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
except Exception as e:
    print(e)
finally:
    # Stop streaming
    pipeline.stop()`

ShettyHarapanahalli, 2020-02-04:
It will be helpful if we have an example for loading json configuration and pixel to point in python.

ev-mp, 2020-02-04:
@ShettyHarapanahalli hello,
The way the RGB and Depth frames are used in the above code is _intuitive and yet profoundly not feasible_ :
The Depth and RGB streams are produced by separate physical sensors (Depth and RGB), using different physical properties, such as aperture size, lenses, sensors. Additionally those sensors are positioned ~1.5cm apart which means that they capture different perspectives. 
Therefore is not possible to obtain the depth of a given RGB pixel by superimposing its coordinates on the Depth frame directly.

I believe that you might misinterpret the colorized depth map in the `realsense-viewer` for the RGB stream, while the colorized depth represents a color-space transition from raw depth (uint16) to RGB (24bit) to make the depth human and visualization-friendly.

In order to achieve the original goal, which in my understanding is picking an RGB pixel and finding its distance, you need to use the `aligned` stream:
- Align streams - [introduction](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#frame-alignment)
- Aligned streams c++ : [demo1](https://github.com/IntelRealSense/librealsense/tree/master/examples/align), [demo2](https://github.com/IntelRealSense/librealsense/tree/master/examples/align-advanced).
- Aligned streams [python example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)

ShettyHarapanahalli, 2020-02-04:
@ev-mp  Thanks for pointing me about align streams, but still I get an error as "'pyrealsense2.pyrealsense2.BufData' object has no attribute 'get_distance'" 

# First import the library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2

# Create a pipeline
pipeline = rs.pipeline()
depth_frame = None

def mouse_drawing(event, x, y, flags, params):
    try:
        if event == cv2.EVENT_LBUTTONDOWN:
            dist = depth_frame.get_distance(int(x), int(y))
            print(f"\r Distance at the selected point x:{x},y:{y} is: {int(dist*1000)} mm")
    except Exception as e:
        print(e)

#Create a config and configure the pipeline to stream
#  different resolutions of color and depth streams
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

# We will be removing the background of objects more than
#  clipping_distance_in_meters meters away
clipping_distance_in_meters = 2 # meters
clipping_distance = clipping_distance_in_meters / depth_scale

# Create an align object
# rs.align allows us to perform alignment of depth frames to others frames
# The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.color
align = rs.align(align_to)
cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
cv2.setMouseCallback("RealSense", mouse_drawing)
# Streaming loop
try:
    while True:
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image

        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue
        depth_frame = aligned_depth_frame.get_data()
        depth_image = np.asanyarray(depth_frame)
        color_image = np.asanyarray(color_frame.get_data())

        # Remove background - Set pixels further than clipping_distance to grey
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)

        # Render images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, depth_colormap))
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', color_image)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()`

ev-mp, 2020-02-04:
@ShettyHarapanahalli , please pay attention to the following:
The mouse callback expect to receive a depth frame
```python
def mouse_drawing(event, x, y, flags, params):
    try:
        if event == cv2.EVENT_LBUTTONDOWN:
            dist = depth_frame.get_distance(int(x), int(y))
            ...
```
While the actual underlying type that you assign to `depth_frame`  is just the pixels payload that is part of the actual frame 
```python
depth_frame = aligned_depth_frame.get_data()
```
The type that the callback expects should be of the type `depth_frame = aligned_depth_frame`.

The link to python API documentation for convenience - https://intelrealsense.github.io/librealsense/python_docs/index.html

ShettyHarapanahalli, 2020-02-05:
Updating the below line woked: 
dist = aligned_depth_frame.get_distance(x, y)

ShettyHarapanahalli, 2020-02-05:
Still not getting correct 3d Coordinates, sharing the function of the same [full code attached as txt file]- 

```python
def point_of_intrest(event, x, y, flags, params):
    try:
        if event == cv2.EVENT_LBUTTONDOWN:
            dist = aligned_depth_frame.get_distance(x, y)
            print(f"\r Distance at the selected point x:{x},y:{y} is: {int(dist * 1000)} mm")
            [x3d,y3d,z3d] = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist*depth_scale)
            print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
            [x3d, y3d, z3d] = rs.rs2_deproject_pixel_to_point(depth_intrin, [x, y], dist * depth_scale)
            print(f"\r 3D coordinates of the selected point using depth are x:{x3d},y:{y3d} and z:{z3d}")
            #print(f"\r 3D coordinates of the selected point are x:{x3d*1000},y:{y3d*1000} and z:{z3d*1000}")
    except Exception as e:
        print(e) 

```python
[3D-Coordinates_of_selected_point.txt](https://github.com/IntelRealSense/librealsense/files/4159474/3D-Coordinates_of_selected_point.txt)
```

ShettyHarapanahalli, 2020-02-05:
Will below code give correct 3d coordinates - 
```python
def point_of_intrest(event, x, y, flags, params):
    try:
        if event == cv2.EVENT_LBUTTONDOWN:
            dist = aligned_depth_frame.get_distance(x, y)
            print(f"\r Distance at the selected point x:{x},y:{y} is: {int(dist * 1000)} mm")
            [x3d,y3d,z3d] = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist*depth_scale)
            print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
            #[x3d, y3d, z3d] = rs.rs2_deproject_pixel_to_point(depth_intrin, [x, y], dist * depth_scale)
            #print(f"\r 3D coordinates of the selected point using depth are x:{x3d},y:{y3d} and z:{z3d}")
    except Exception as e:
        print(e)
```
Full code attached as txt file. 
[3D-Coordinates_of_selected_point.txt](https://github.com/IntelRealSense/librealsense/files/4159546/3D-Coordinates_of_selected_point.txt)

ShettyHarapanahalli, 2020-02-05:
With the above code, I get an error as 
"module 'pyrealsense2' has no attribute 'rs2_translate_point_to_point'"

Not sure if we have this interface in python, I was not able to get any references for "rs2_translate_point_to_point" and "point_to_point" from the link shared above.  Please suggest the correct api and sequence to do.

Sharing full code for reference. 
[3D-Coordinates_of_selected_point.txt](https://github.com/IntelRealSense/librealsense/files/4160812/3D-Coordinates_of_selected_point.txt)

ev-mp, 2020-02-11:
@ShettyHarapanahalli , the function you're looking for is [`rs2_transform_point_to_point`]( 
https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L126)

More on the topic of 2D and 3D correlation can be found in https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

ShettyHarapanahalli, 2020-02-11:
@ev-mp the Z values from get distance and transform_point_to_point do not match. (sharing the output for reference).    

Depth Scale is:  0.0010000000474974513

 Distance at the selected pixel x:501,y:370 is: **2.820000171661377** mtr
 3D coordinates of the corresponding selected pixel x:0.0008112132200039923,y:0.0005335451569408178 and z:**0.002820000285282731**

 Distance at the selected pixel x:501,y:370 is: **2.820000171661377** mtr
 3D coordinates of the corresponding selected pixel x:0.0008112132200039923,y:0.0005335451569408178 and z:**0.002820000285282731**

 Distance at the selected pixel x:502,y:369 is: **2.7940001487731934** mtr
 3D coordinates of the corresponding selected pixel x:0.00080829334910959,y:0.0005240681348368526 and z:**0.002794000320136547**

[3D-Coordinates_of_selected_point.txt](https://github.com/IntelRealSense/librealsense/files/4184923/3D-Coordinates_of_selected_point.txt)

ev-mp, 2020-02-11:
@ShettyHarapanahalli , `rs2_deproject_pixel_to_point` accepts depth in metric unit. Apply the following to correct:
>dist = aligned_depth_frame.get_distance(x, y)
            print(f"\r Distance at the selected pixel x:{x},y:{y} is: {dist} **mtr**")
            depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],**dist**~~*depth_scale~~)
            [x3d,y3d,z3d] = rs.rs2_transform_point_to_point(depth_to_color_extrin,depth_point)
```

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@ShettyHarapanahalli Do you have any other questions with the reply from ev-mp? Looking forward to your update. Thanks!

closed 2020-02-17 by ShettyHarapanahalli

srividya91, 2020-02-18:
@ShettyHarapanahalli Thanks for the response.Actually i am trying to calculate the distance for the detected object to depth camera.. so i am trying to add your code to my object deetction.. but i am getting run time error.

here is my source code
`import cv2 as cv
import cv2
import numpy as np
import pyrealsense2 as rs
import os
import time
import matplotlib.pyplot as plt
import imutils
from imutils import paths
#Write down conf, nms thresholds,inp width/height
confThreshold = 0.50
nmsThreshold = 0.40
inpWidth = 416
inpHeight = 416
#Load names of classes and turn that into a list
classesFile=r'C:\Users\vistan\Desktop\files\coco.names'
classes = None
with open(classesFile,'rt') as f:
    classes = f.read().rstrip('\n').split('\n')
#Model configuration
modelConf = r'C:\Users\vistan\Desktop\files\yolov3.cfg'
modelWeights = r'C:\Users\vistan\Desktop\files\yolov3.weights'
# modelConf='C:\\Users\\poornachandra\\Desktop\\originals@9\\darknet-master\\darknet-master\\cfg\\yolo9000.cfg'
# modelWeights='C:\\Users\\poornachandra\\Desktop\\originals@9\\yolo9000.weights'
k=[]
def postprocess(images, outs):
    frameHeight = images.shape[0]
    frameWidth = images.shape[1]
    classIDs = []
    confidences = []
    boxes = []
    for out in outs:
        for detection in out:
            scores = detection [5:]
            classID = np.argmax(scores)
            confidence = scores[classID]
            if confidence > confThreshold:
                centerX = int(detection[0] * frameWidth)
                centerY = int(detection[1] * frameHeight)
                width = int(detection[2]* frameWidth)
                height = int(detection[3]*frameHeight )
                left = int(centerX - width/2)
                top = int(centerY - height/2)
                classIDs.append(classID)
                confidences.append(float(confidence))
                boxes.append([left, top, width, height])
    indices = cv.dnn.NMSBoxes (boxes,confidences, confThreshold, nmsThreshold )
    indices = cv.dnn.NMSBoxes(boxes, confidences, confThreshold, nmsThreshold)
    for i in indices:
        i = i[0]
        box = boxes[i]
        left = box[0]
        top = box[1]
        width = box[2]
        height = box[3]
        drawPred(classIDs[i], confidences[i], left, top, left + width, top + height)
def drawPred(classId, conf, left, top, right, bottom,):
    # Draw a bounding box.
    cv.rectangle(images, (left, top), (right, bottom), (255, 178, 50), 3)
    label = '%.2f' % conf
    # Get the label for the class name and its confidence
    if classes:
        assert (classId < len(classes))
        label = '%s:%s' % (classes[classId], label)
             
    **dist = aligned_depth_frame.get_distance(left,top)
    print(dist)

    print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")
    depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist)
    [x3d,y3d,z3d] = rs.rs2_translate_point_to_point(depth_to_color_extrin,depth_point)
    print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
    print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")**
    
   cv.putText(images,label,(left, top), cv.FONT_HERSHEY_SIMPLEX, 1, (255, 255, 255), 3)
    
def getOutputsNames(net):
    # Get the names of all the layers in the network
    layersNames = net.getLayerNames()
    # Get the names of the output layers, i.e. the layers with unconnected outputs
    return [layersNames[i[0] - 1] for i in net.getUnconnectedOutLayers()]
#Set up the net
net = cv.dnn.readNetFromDarknet(modelConf, modelWeights)
net.setPreferableBackend(cv.dnn.DNN_BACKEND_OPENCV)
net.setPreferableTarget(cv.dnn.DNN_TARGET_CPU)

pipe=rs.pipeline()
cfg=rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#start streaming
profile=pipe.start(cfg)
#getting the depth sensors depth scale
depth_sensor=profile.get_device().first_depth_sensor()
depth_scale=depth_sensor.get_depth_scale()

align_to=rs.stream.color
align=rs.align(align_to)
while True:
    #get frame from video
    #hasFrame, frame = cap.read()
    frames=pipe.wait_for_frames()
    
    #align the depth frame to color frame
    aligned_frames=align.process(frames)
    
    #get aligned frames
    aligned_depth_frame=aligned_frames.get_depth_frame()
    color_frame=aligned_frames.get_color_frame()
    
    #validate that both frames are valid
    if not aligned_depth_frame or not color_frame:
        continue
   
    width = aligned_depth_frame.get_width()
    height = aligned_depth_frame.get_height()
    
    depth_image=np.asanyarray(aligned_depth_frame.get_data())
    color_image=np.asanyarray(color_frame.get_data())
    
    #Render color images
    depth_color_map=cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
    images = np.hstack((color_image, depth_color_map))
    blob = cv.dnn.blobFromImage(images, 1/255, (inpWidth, inpHeight), [0,0,0], 1, crop = False)
    #Set the input the the net
    net.setInput(blob)
    outs = net.forward (getOutputsNames(net))
    postprocess (images, outs)
    
    #show the image
    cv2.namedWindow('Object Detection with openCV', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('Object Detection with openCV', images)
    key = cv2.waitKey(1)
    # Press esc or 'q' to close the image window
    if key & 0xFF == ord('q') or key == 27:
        cv2.destroyAllWindows()
        break

`

srividya91, 2020-02-18:
@ShettyHarapanahalli Thanks for the response.Actually i am trying to calculate the distance for the detected object to depth camera.. so i am trying to add your code to my object deetction.. but i am getting run time error.
here is my source code
`
import cv2 as cv
import cv2
import numpy as np
import pyrealsense2 as rs
import os
import time
import matplotlib.pyplot as plt
import imutils
from imutils import paths
#Write down conf, nms thresholds,inp width/height
confThreshold = 0.50
nmsThreshold = 0.40
inpWidth = 416
inpHeight = 416
#Load names of classes and turn that into a list
classesFile=r'C:\Users\vistan\Desktop\files\coco.names'
classes = None
with open(classesFile,'rt') as f:
classes = f.read().rstrip('\n').split('\n')
#Model configuration
modelConf = r'C:\Users\vistan\Desktop\files\yolov3.cfg'
modelWeights = r'C:\Users\vistan\Desktop\files\yolov3.weights'

k=[]
def postprocess(images, outs):
frameHeight = images.shape[0]
frameWidth = images.shape[1]
classIDs = []
confidences = []
boxes = []
for out in outs:
for detection in out:
scores = detection [5:]
classID = np.argmax(scores)
confidence = scores[classID]
if confidence > confThreshold:
centerX = int(detection[0] * frameWidth)
centerY = int(detection[1] * frameHeight)
width = int(detection[2]* frameWidth)
height = int(detection[3]*frameHeight )
left = int(centerX - width/2)
top = int(centerY - height/2)
classIDs.append(classID)
confidences.append(float(confidence))
boxes.append([left, top, width, height])
indices = cv.dnn.NMSBoxes (boxes,confidences, confThreshold, nmsThreshold )
indices = cv.dnn.NMSBoxes(boxes, confidences, confThreshold, nmsThreshold)
for i in indices:
i = i[0]
box = boxes[i]
left = box[0]
top = box[1]
width = box[2]
height = box[3]
drawPred(classIDs[i], confidences[i], left, top, left + width, top + height)
def drawPred(classId, conf, left, top, right, bottom,):
# Draw a bounding box.
cv.rectangle(images, (left, top), (right, bottom), (255, 178, 50), 3)
label = '%.2f' % conf
# Get the label for the class name and its confidence
if classes:
assert (classId < len(classes))
label = '%s:%s' % (classes[classId], label)

**dist = aligned_depth_frame.get_distance(left,top)
print(dist)

print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")
depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist)
[x3d,y3d,z3d] = rs.rs2_translate_point_to_point(depth_to_color_extrin,depth_point)
print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")**
cv.putText(images,label,(left, top), cv.FONT_HERSHEY_SIMPLEX, 1, (255, 255, 255), 3)

def getOutputsNames(net):
# Get the names of all the layers in the network
layersNames = net.getLayerNames()
# Get the names of the output layers, i.e. the layers with unconnected outputs
return [layersNames[i[0] - 1] for i in net.getUnconnectedOutLayers()]
#Set up the net
net = cv.dnn.readNetFromDarknet(modelConf, modelWeights)
net.setPreferableBackend(cv.dnn.DNN_BACKEND_OPENCV)
net.setPreferableTarget(cv.dnn.DNN_TARGET_CPU)

pipe=rs.pipeline()
cfg=rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#start streaming
profile=pipe.start(cfg)
#getting the depth sensors depth scale
depth_sensor=profile.get_device().first_depth_sensor()
depth_scale=depth_sensor.get_depth_scale()

align_to=rs.stream.color
align=rs.align(align_to)
while True:
#get frame from video
#hasFrame, frame = cap.read()
frames=pipe.wait_for_frames()

#align the depth frame to color frame
aligned_frames=align.process(frames)

#get aligned frames
aligned_depth_frame=aligned_frames.get_depth_frame()
color_frame=aligned_frames.get_color_frame()

#validate that both frames are valid
if not aligned_depth_frame or not color_frame:
    continue

width = aligned_depth_frame.get_width()
height = aligned_depth_frame.get_height()

depth_image=np.asanyarray(aligned_depth_frame.get_data())
color_image=np.asanyarray(color_frame.get_data())

#Render color images
depth_color_map=cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
images = np.hstack((color_image, depth_color_map))
blob = cv.dnn.blobFromImage(images, 1/255, (inpWidth, inpHeight), [0,0,0], 1, crop = False)
#Set the input the the net
net.setInput(blob)
outs = net.forward (getOutputsNames(net))
postprocess (images, outs)

#show the image
cv2.namedWindow('Object Detection with openCV', cv2.WINDOW_AUTOSIZE)
cv2.imshow('Object Detection with openCV', images)
key = cv2.waitKey(1)
# Press esc or 'q' to close the image window
if key & 0xFF == ord('q') or key == 27:
    cv2.destroyAllWindows()
    break
`

srividya91, 2020-02-18:
@ShettyHarapanahalli Thanks for the response.Actually i am trying to calculate the distance for the detected object to depth camera.. so i am trying to add your code to my object deetction.. but i am getting run time error.

here is my source code
`import cv2 as cv
import cv2
import numpy as np
import pyrealsense2 as rs
import os
import time
import matplotlib.pyplot as plt
import imutils
from imutils import paths

confThreshold = 0.50
nmsThreshold = 0.40
inpWidth = 416
inpHeight = 416

classesFile=r'C:\Users\vistan\Desktop\files\coco.names'
classes = None
with open(classesFile,'rt') as f:
classes = f.read().rstrip('\n').split('\n')

modelConf = r'C:\Users\vistan\Desktop\files\yolov3.cfg'
modelWeights = r'C:\Users\vistan\Desktop\files\yolov3.weights'

k=[]
def postprocess(images, outs):
frameHeight = images.shape[0]
frameWidth = images.shape[1]
classIDs = []
confidences = []
boxes = []
for out in outs:
for detection in out:
scores = detection [5:]
classID = np.argmax(scores)
confidence = scores[classID]
if confidence > confThreshold:
centerX = int(detection[0] * frameWidth)
centerY = int(detection[1] * frameHeight)
width = int(detection[2]* frameWidth)
height = int(detection[3]*frameHeight )
left = int(centerX - width/2)
top = int(centerY - height/2)
classIDs.append(classID)
confidences.append(float(confidence))
boxes.append([left, top, width, height])
indices = cv.dnn.NMSBoxes (boxes,confidences, confThreshold, nmsThreshold )
indices = cv.dnn.NMSBoxes(boxes, confidences, confThreshold, nmsThreshold)
for i in indices:
i = i[0]
box = boxes[i]
left = box[0]
top = box[1]
width = box[2]
height = box[3]
drawPred(classIDs[i], confidences[i], left, top, left + width, top + height)
def drawPred(classId, conf, left, top, right, bottom,):

cv.rectangle(images, (left, top), (right, bottom), (255, 178, 50), 3)
label = '%.2f' % conf

if classes:
assert (classId < len(classes))
label = '%s:%s' % (classes[classId], label)

**dist = aligned_depth_frame.get_distance(left,top)
print(dist)

print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")
depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist)
[x3d,y3d,z3d] = rs.rs2_translate_point_to_point(depth_to_color_extrin,depth_point)
print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")**
cv.putText(images,label,(left, top), cv.FONT_HERSHEY_SIMPLEX, 1, (255, 255, 255), 3)

def getOutputsNames(net):
# Get the names of all the layers in the network
layersNames = net.getLayerNames()
# Get the names of the output layers, i.e. the layers with unconnected outputs
return [layersNames[i[0] - 1] for i in net.getUnconnectedOutLayers()]
#Set up the net
net = cv.dnn.readNetFromDarknet(modelConf, modelWeights)
net.setPreferableBackend(cv.dnn.DNN_BACKEND_OPENCV)
net.setPreferableTarget(cv.dnn.DNN_TARGET_CPU)

pipe=rs.pipeline()
cfg=rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#start streaming
profile=pipe.start(cfg)
#getting the depth sensors depth scale
depth_sensor=profile.get_device().first_depth_sensor()
depth_scale=depth_sensor.get_depth_scale()

align_to=rs.stream.color
align=rs.align(align_to)
while True:
#get frame from video
#hasFrame, frame = cap.read()
frames=pipe.wait_for_frames()

#align the depth frame to color frame
aligned_frames=align.process(frames)

#get aligned frames
aligned_depth_frame=aligned_frames.get_depth_frame()
color_frame=aligned_frames.get_color_frame()

#validate that both frames are valid
if not aligned_depth_frame or not color_frame:
    continue

width = aligned_depth_frame.get_width()
height = aligned_depth_frame.get_height()

depth_image=np.asanyarray(aligned_depth_frame.get_data())
color_image=np.asanyarray(color_frame.get_data())

#Render color images
depth_color_map=cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
images = np.hstack((color_image, depth_color_map))
blob = cv.dnn.blobFromImage(images, 1/255, (inpWidth, inpHeight), [0,0,0], 1, crop = False)
#Set the input the the net
net.setInput(blob)
outs = net.forward (getOutputsNames(net))
postprocess (images, outs)

#show the image
cv2.namedWindow('Object Detection with openCV', cv2.WINDOW_AUTOSIZE)
cv2.imshow('Object Detection with openCV', images)
key = cv2.waitKey(1)
# Press esc or 'q' to close the image window
if key & 0xFF == ord('q') or key == 27:
    cv2.destroyAllWindows()
    break

srividya91, 2020-02-18:
@ShettyHarapanahalli Thanks for the response.Actually i am trying to calculate the distance for the detected object to depth camera.. so i am trying to add your code to my object deetction.. but i am getting run time error.

here is my source code
`import cv2 as cv
import cv2
import numpy as np
import pyrealsense2 as rs
import os
import time
import matplotlib.pyplot as plt
import imutils
from imutils import paths

confThreshold = 0.50
nmsThreshold = 0.40
inpWidth = 416
inpHeight = 416

classesFile=r'C:\Users\vistan\Desktop\files\coco.names'
classes = None
with open(classesFile,'rt') as f:
classes = f.read().rstrip('\n').split('\n')

modelConf = r'C:\Users\vistan\Desktop\files\yolov3.cfg'
modelWeights = r'C:\Users\vistan\Desktop\files\yolov3.weights'

k=[]
def postprocess(images, outs):
frameHeight = images.shape[0]
frameWidth = images.shape[1]
classIDs = []
confidences = []
boxes = []
for out in outs:
for detection in out:
scores = detection [5:]
classID = np.argmax(scores)
confidence = scores[classID]
if confidence > confThreshold:
centerX = int(detection[0] * frameWidth)
centerY = int(detection[1] * frameHeight)
width = int(detection[2]* frameWidth)
height = int(detection[3]*frameHeight )
left = int(centerX - width/2)
top = int(centerY - height/2)
classIDs.append(classID)
confidences.append(float(confidence))
boxes.append([left, top, width, height])
indices = cv.dnn.NMSBoxes (boxes,confidences, confThreshold, nmsThreshold )
indices = cv.dnn.NMSBoxes(boxes, confidences, confThreshold, nmsThreshold)
for i in indices:
i = i[0]
box = boxes[i]
left = box[0]
top = box[1]
width = box[2]
height = box[3]
drawPred(classIDs[i], confidences[i], left, top, left + width, top + height)
def drawPred(classId, conf, left, top, right, bottom,):

cv.rectangle(images, (left, top), (right, bottom), (255, 178, 50), 3)
label = '%.2f' % conf

if classes:
assert (classId < len(classes))
label = '%s:%s' % (classes[classId], label)

_
dist = aligned_depth_frame.get_distance(left,top)
print(dist)

print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")
depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist)
[x3d,y3d,z3d] = rs.rs2_translate_point_to_point(depth_to_color_extrin,depth_point)
print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")_
cv.putText(images,label,(left, top), cv.FONT_HERSHEY_SIMPLEX, 1, (255, 255, 255), 3)

def getOutputsNames(net):
# Get the names of all the layers in the network
layersNames = net.getLayerNames()
# Get the names of the output layers, i.e. the layers with unconnected outputs
return [layersNames[i[0] - 1] for i in net.getUnconnectedOutLayers()]
#Set up the net
net = cv.dnn.readNetFromDarknet(modelConf, modelWeights)
net.setPreferableBackend(cv.dnn.DNN_BACKEND_OPENCV)
net.setPreferableTarget(cv.dnn.DNN_TARGET_CPU)

pipe=rs.pipeline()
cfg=rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#start streaming
profile=pipe.start(cfg)
#getting the depth sensors depth scale
depth_sensor=profile.get_device().first_depth_sensor()
depth_scale=depth_sensor.get_depth_scale()

align_to=rs.stream.color
align=rs.align(align_to)
while True:
#get frame from video
#hasFrame, frame = cap.read()
frames=pipe.wait_for_frames()

#align the depth frame to color frame
aligned_frames=align.process(frames)

#get aligned frames
aligned_depth_frame=aligned_frames.get_depth_frame()
color_frame=aligned_frames.get_color_frame()

#validate that both frames are valid
if not aligned_depth_frame or not color_frame:
    continue

width = aligned_depth_frame.get_width()
height = aligned_depth_frame.get_height()

depth_image=np.asanyarray(aligned_depth_frame.get_data())
color_image=np.asanyarray(color_frame.get_data())

#Render color images
depth_color_map=cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
images = np.hstack((color_image, depth_color_map))
blob = cv.dnn.blobFromImage(images, 1/255, (inpWidth, inpHeight), [0,0,0], 1, crop = False)
#Set the input the the net
net.setInput(blob)
outs = net.forward (getOutputsNames(net))
postprocess (images, outs)

#show the image
cv2.namedWindow('Object Detection with openCV', cv2.WINDOW_AUTOSIZE)
cv2.imshow('Object Detection with openCV', images)
key = cv2.waitKey(1)
# Press esc or 'q' to close the image window
if key & 0xFF == ord('q') or key == 27:
    cv2.destroyAllWindows()
    break

srividya91, 2020-02-18:
@ShettyHarapanahalli Thanks for the response.Actually i am trying to calculate the distance for the detected object to depth camera.. so i am trying to add your code to my object deetction.. but i am getting run time error.

here is my source code
`import cv2 as cv
import cv2
import numpy as np
import pyrealsense2 as rs
import os
import time
import matplotlib.pyplot as plt
import imutils
from imutils import paths
#Write down conf, nms thresholds,inp width/height
confThreshold = 0.50
nmsThreshold = 0.40
inpWidth = 416
inpHeight = 416
#Load names of classes and turn that into a list
#classesFile = "C:\\Users\\poornachandra\\Desktop\\DS-PROJECTS\\FC-OBJ\\coco.names"
classesFile=r'C:\Users\vistan\Desktop\files\coco.names'
#classesFile='C:\\Users\\poornachandra\\Desktop\\originals@9\\darknet-master\\darknet-master\\data\\9k.names'
classes = None
with open(classesFile,'rt') as f:
    classes = f.read().rstrip('\n').split('\n')
#Model configuration
modelConf = r'C:\Users\vistan\Desktop\files\yolov3.cfg'
modelWeights = r'C:\Users\vistan\Desktop\files\yolov3.weights'

k=[]

def postprocess(images, outs):
    frameHeight = images.shape[0]
    frameWidth = images.shape[1]
    classIDs = []
    confidences = []
    boxes = []
    for out in outs:
        for detection in out:
            scores = detection [5:]
            classID = np.argmax(scores)
            confidence = scores[classID]
            if confidence > confThreshold:
                centerX = int(detection[0] * frameWidth)
                centerY = int(detection[1] * frameHeight)
                width = int(detection[2]* frameWidth)
                height = int(detection[3]*frameHeight )
                left = int(centerX - width/2)
                top = int(centerY - height/2)
                classIDs.append(classID)
                confidences.append(float(confidence))
                boxes.append([left, top, width, height])
    indices = cv.dnn.NMSBoxes (boxes,confidences, confThreshold, nmsThreshold )
    indices = cv.dnn.NMSBoxes(boxes, confidences, confThreshold, nmsThreshold)
    for i in indices:
        i = i[0]
        box = boxes[i]
        left = box[0]
        top = box[1]
        width = box[2]
        height = box[3]
        drawPred(classIDs[i], confidences[i], left, top, left + width, top + height)
def drawPred(classId, conf, left, top, right, bottom,):
    # Draw a bounding box.
    cv.rectangle(images, (left, top), (right, bottom), (255, 178, 50), 3)
    label = '%.2f' % conf
    # Get the label for the class name and its confidence
    if classes:
        assert (classId < len(classes))
        label = '%s:%s' % (classes[classId], label)
             
    _dist = aligned_depth_frame.get_distance(left,top)
    print(dist)

    print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")
    depth_point = rs.rs2_deproject_pixel_to_point(color_intrin,[x,y],dist)
    [x3d,y3d,z3d] = rs.rs2_translate_point_to_point(depth_to_color_extrin,depth_point)
    print(f"\r 3D coordinates of the selected point using color are x:{x3d},y:{y3d} and z:{z3d}")
    print(f"\r Distance at the selected point x:{left},y:{top} is: {int(dist * 1000)} mm")_
    
    cv.putText(images,label,(left, top), cv.FONT_HERSHEY_SIMPLEX, 1, (255, 255, 255), 3)
    #cv.putText(frame, pr, (left+1,top+10),  2, (255, 255, 255), 3)
def getOutputsNames(net):
    # Get the names of all the layers in the network
    layersNames = net.getLayerNames()
    # Get the names of the output layers, i.e. the layers with unconnected outputs
    return [layersNames[i[0] - 1] for i in net.getUnconnectedOutLayers()]
#Set up the net
net = cv.dnn.readNetFromDarknet(modelConf, modelWeights)
net.setPreferableBackend(cv.dnn.DNN_BACKEND_OPENCV)
net.setPreferableTarget(cv.dnn.DNN_TARGET_CPU)

pipe=rs.pipeline()
cfg=rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
#start streaming
profile=pipe.start(cfg)
#getting the depth sensors depth scale
depth_sensor=profile.get_device().first_depth_sensor()
depth_scale=depth_sensor.get_depth_scale()

align_to=rs.stream.color
align=rs.align(align_to)
while True:
    #get frame from video
    #hasFrame, frame = cap.read()
    frames=pipe.wait_for_frames()
    #align the depth frame to color frame
    aligned_frames=align.process(frames)
    #get aligned frames
    aligned_depth_frame=aligned_frames.get_depth_frame()
    color_frame=aligned_frames.get_color_frame()
    #validate that both frames are valid
    if not aligned_depth_frame or not color_frame:
        continue
    width = aligned_depth_frame.get_width()
    height = aligned_depth_frame.get_height()
    depth_image=np.asanyarray(aligned_depth_frame.get_data())
    color_image=np.asanyarray(color_frame.get_data())
    #Render color images
depth_color_map=cv2.applyColorMap(cv2.convertScaleAbs(depth_image,alpha=0.03),cv2.COLORMAP_JET)
    images = np.hstack((color_image, depth_color_map))
    #postprocess (images, outs)
    #if hasFrame==True:
    #Create a 4D blob from a frame
    blob = cv.dnn.blobFromImage(images, 1/255, (inpWidth, inpHeight), [0,0,0], 1, crop = False)
    #Set the input the the net
    net.setInput(blob)
    outs = net.forward (getOutputsNames(net))
    postprocess (images, outs)
    #show the image
    cv2.namedWindow('Object Detection with openCV', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('Object Detection with openCV', images)
    key = cv2.waitKey(1)
    # Press esc or 'q' to close the image window
    if key & 0xFF == ord('q') or key == 27:
        cv2.destroyAllWindows()
        break`

    
    
`

srividya91, 2020-02-19:
@ShettyHarapanahalli with the above code. i got an error as 

"rs2_deproject_pixel_to_point(): incompatible function arguments.
 The following argument types are supported:
    1. (intrin: pyrealsense2.pyrealsense2.intrinsics, pixel: List[float[2]], depth: float) -> List[float[3]]"

please tell me how to pass the arguments.