realsenseai/librealsense #5781: Reboot USB connection on windows

Allius27, 2020-02-04:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|915112071095|
|**Firmware Version**|05.11.15.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1, 3.1|
|**Product Line**|D400|
|**Asic Serial Number**|901523021687|
|**Firmware Update Id**|901523021687|

Good day.
Sometimes after unplug and plug in device to USB3.0 camera might determine as USB2.0 in librealsense , and images with this kind of connection is not usable for analize.

Viewer print message "USB SCP overflow"

Do you know any way to reset/reboot usb port?

Here examle of problem infrared image, which was got through USB2.0 OS connection (but phisically device plugged to USB3.0). ( Image is too dark )

[image]

MartyG-RealSense, 2020-02-04:
You can reset the camera in the RealSense Viewer by going to the 'More' option at the top of the options side-panel (the icon that looks like a pile of plates') and selecting 'Hardware Reset' from its menu.  

[image]

A reset can also be performed with scripting using the **hardware_reset()** instruction.  The link below has a C++ example:

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

Allius27, 2020-02-05:
Thank you for reply.

I have tested this functional with one Viewer, and it works perfect. After disconnect camera have always detected as USB3.

But in my code, when I unplug camera and plug it again, somethimes it might determine as USB2.

I have found an example, when Viewer can determine it as USB2.0. Two processes of Viewer must be execute at the same time. And hardware reset doesn't work in this case.

Gif below
[image]

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

As to the dark IR image issue, please try enable the  the auto-exposure
based on the capture screen you attached, it looks be disabled
or you can increase exposure time of IR camera that should definitely can help the IR image brighter

Stereo Module->Control->Exposure

with USB2 connection, the auto-exposure feature should be still supported.
Basically, compared with USB3
-USB2 has smaller data bandwidth that not able to support high resolution with high fps
that is the reason easier to see the buffer overflow issue
-USB2 VBUS has smaller power can be delivered to device 900mA (USB3)  vs 500mA (USB2)

By the way, turn on the projector should help to improve the depth image fill rate and accuracy

Allius27, 2020-02-05:
I think, that problem have resolved for this question.

Obviously, this problem belongs to camera version **05.11.15.00**
I have updated firmware for camera up to **05.12.02.100**, and problem was solved

With new software version independently on USB connection (USB2 or USB3) IR image is always perfect

Allius27, 2020-02-05:
So, I hurried to close this issue.

My question is:
Is there some way to change connection from USB2 to USB3 on windows (Hardware reset doesn't work)?
May be without using librealsense library

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

Hi Allius27
The hardware reset should able to change the USB2 connection to USB3 connection if there is no USB3 signal integrity issue
there do some way to disable/enable the USB port that cause USB port re-enumerated the device that you can find on the website
but in your case, it looks the after hardware reset command is issued,
host try to enumerated device as USB3 still fail, then host try to enumerated device as USB2 instead...
that means even you use other approach to reset the USB port for re-enumeration, you may still face the same problem

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

Hi Allius27,
had you found the solution work for you?
by the way. you mentioned that sometimes the device is recognized as USB2 device when you unplug then plug the camera agin this should be able to be fixed if you plug the USB3 connector faster to the host USB3 port, or doing hw-reset should help

Allius27, 2020-03-20:
Hi. 
No, I don't find any solution for this problem.
And gif above shows that hardware reset don't work in my case

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi Allius27,
Any chance to change the cable to see if it help?
please refer to
https://github.com/IntelRealSense/librealsense/issues/4105

RealSenseCustomerSupport, 2020-04-20:
----------------------------------------------

Hi Allius27,
Any chance to change the cable to see if it help?
please refer to
#4105
or could you change to different PC to see issue still there
sometimes the host USB3 port + cable 's signal integrity would cause this issue

closed 2020-07-28 by RealSenseCustomerSupport