realsenseai/librealsense #5782: Power issue on raspberry pi with two SR305 cameras

piraog, 2020-02-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |  Raspbian GNU/Linux 10 (buster)  | 
| Kernel Version (Linux Only)    |  4.19.75-v7l+                                      | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2.32.1                         | 
| Language                            |  python                        | 
| Segment			|  others                          | 

### Issue Description
Hello,
I have a problem with my current project. I'm trying to use 2 SR305 cameras on the same RPI 4. I need to use both infrared streams for now (no depth or RGB).

When using the realsense viewer, I have no trouble getting both infrared streams to display.

In my python code, I use the realsense_device_manager from the box_dimensioner_multicam example. It is working fine with one camera, and no problem running it on my windows laptop with 2 sr305. But when starting the pipeline with two SR305 on the RPI, I get the following error:

Traceback (most recent call last):
  File "visualize.py", line 42, in <module>
    device_manager.enable_all_devices()
  File "/home/pi/formeuse/realsense_device_manager.py", line 159, in enable_all_devices
    self.enable_device(serial, enable_ir_emitter)
  File "/home/pi/formeuse/realsense_device_manager.py", line 144, in enable_device
    pipeline_profile = pipeline.start(self._config)
RuntimeError: failed to set power state

I initially thought it was because the pi usb hub couldn't deliver enough power, but since the viewer is doing it just fine, I must be missing something.

Thanks in advance for your help :)

ev-mp, 2020-02-04:
@piraog , the `realsense_device_manager.py` uses the camera's default configuration (`self._config`)
that for SR305 comprise of Depth+RGB streams. So unless you've overwritten the default configuration trying to start two devices would result in activation of 4 streams which draws more power than RPi4 USB ports can sustain.

piraog, 2020-02-05:
@ev-mp Thanks for the quick answer
I did try to overwrite the realsense_device_manager, but somehow dodn't manage to do it completly.
In the end, without using the device manager It ended up working.

Thanks again for the quick reply

closed 2020-02-05 by piraog