realsenseai/librealsense #5783: 100% CPU Usage

radfordi, 2020-02-04:
Using 83f952a and a D435i w/ `5.12.2.100`, I see 100% CPU usage with this program (on a Mac):
```python
import pyrealsense2 as rs
c = rs.context()
d = c.devices[0]
import time
time.sleep(100)
```

timwedde, 2020-03-17:
I can confirm this issue as well.
Firmware: `05.12.03.00`
librealsense and pyrealsense2: `2.33.1` (compiled today from latest `master` at 1d7ef6ac04394d6183178f6e2dd9304d34bb44ff)
Hardware: `D415`

Tested on both my MacBook Pro Late 2014 (i7 4870HQ) as well as my iMac with a 9900K at 5.1GHz in it. I find it quite strange that the process takes up 100% of a core on both, given just how much more powerful the 9900K should be.
As such I suspect something is spinning empty somewhere in librealsense and just taking up as many cycles as it can.

librealsense was compiled with OpenMP `off` as it was mentioned in some other issues that this may affect it. Alas, it did not.

Something else that I found in experimenting with the Python version (though I do not know whether that is useful or not, but I mention it just in case) is that this behavior only occurs as long as the reference is held in the test program. As soon as the reference is gone, the CPU usage drops to basically 0%.

```python
import time
import pyrealsense2 as rs

c = rs.context()
d = c.devices[0]  # CPU usage spikes here

time.sleep(5)

del d  # CPU usage drops here

time.sleep(95)
```

Is this intended behavior?

radfordi, 2020-03-17:
The fact that `del d` drops the CPU is expected.  Likely device creates threads that spin and when you delete the device, the threads get cleaned up.

timwedde, 2020-03-18:
This fix worked perfectly for me. CPU usage is now in line with what I'm seeing on Ubuntu.
Great work 👍

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

Thanks for sharing!
If there no further discussion in 7 days, this issue will be closed

PR #6083 (2020-03-25): Fix 100% CPU usage when using libusb
Replace broken dispatcher with a thread to fix 100% CPU usage

The dispatcher can't tell the difference between "timeout" and "empty", so it ends up infinte looping.  Using a thread directly is much simpler and more correct.

Fixes #5783.  Fixes #6062.
closed 2020-04-04 by RealSenseCustomerSupport