realsenseai/librealsense #5784: Converting numpy depth_map back into pyrealsense2 depth frame

LMurphy99, 2020-02-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i} | 
| Firmware Version                   | (N/A) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC |
| SDK Version                            |  {N/A }                          | 
| Language                            |  {python }                          | 
| Segment			|  {N/A}                          | 

### Issue Description
I have been working on pre-processing depth maps before I convert them to pointclouds. This involved converting the depth_frame.get_data() into a numpy array, and doing some processing (i.e, converting all pixels of no interest to nan). I wanted to use the rs.pointcloud().calculate() function to calculate my point cloud, with my altered depth map as input. I'm getting the following error however.. 

"TypeError: calculate(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.pointcloud, depth: pyrealsense2.pyrealsense2.frame) -> pyrealsense2.pyrealsense2.points"

i was wondering how I would go about converting my numpy depth map into a format that the function can accept.

Attached is the code of note
[image]

zapaishchykova, 2020-04-05:
following

rcroset, 2020-04-14:
Interested in an answer too !

r-ratajczak, 2020-04-20:
Hi,
I also try to convert back the numpy array to BufData in order to reuse the modified data. I tried casting, memoryview, creating new BufData objects but without any success. Is there any possibility to recreate BufData object? The environment returns an error, that there is no defined constructor.

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

@LMurphy99 Sorry for late response. Have submitted your requirement to engineering team. Any update will let you know. Thanks!

jts3k, 2020-08-08:
following

adityashrm21, 2020-10-23:
Are there any updates on this? Thanks!

RealSenseSupport, 2020-10-27:
@adityashrm21 Sorry, this enhancement is still under working. Any update will let you know. Thanks!

tsaizhenling, 2020-10-29:
following

MartyG-RealSense, 2021-02-22:
Hi everyone,

The BufData object creation question has been answered by a RealSense team member in the link below.

https://github.com/IntelRealSense/librealsense/issues/8394#issuecomment-782811459

kailkuhn, 2021-03-07:
@MartyG-RealSense is there any updates to this? thanks

MartyG-RealSense, 2021-03-07:
Hi @kailkuhn  If you are asking about the official feature request for numpy depth map conversion, there are no updates to report.

There have been RealSense users who have wanted to convert a numpy array to a depth frame, or a numpy to a point cloud. 
 As far as I know there is not a finished solution available yet though, but the information about their own attempts may provide you with some useful insights.

https://github.com/IntelRealSense/librealsense/issues/5784 

https://github.com/IntelRealSense/librealsense/issues/2551

DeepakChandra2612, 2021-11-03:
Hi All,

Any updates on this feature to convert the numpy array back to real sense depth frames to utilize the inbuilt functions of pyrealsense2?

Thanks

MartyG-RealSense, 2021-11-03:
Hi @DeepakChandra2612  There is no further information to report about the official feature request to Intel, though it remains active.  I am not aware of any further progress that RealSense users have made themselves on developing numpy to rs2::frame conversion either since the https://github.com/IntelRealSense/librealsense/issues/2551 link already mentioned.

sanjaiiv04, 2023-11-20:
I am trying to get a region of the depth map and convert it into point cloud. To do so, I have cropped the depth map to get the ROI I wanted and in doing so, the depth frame was converted into a numpy array. I want to convert this numpy array back to depth frame format so that I can extract point clouds from the cropped region. How to do that? Here is the code:

pipeline = rs.pipeline()
config = rs.config()
colorizer = rs.colorizer()
dec_filter = rs.decimation_filter ()   # Decimation - reduces depth frame density
spat_filter = rs.spatial_filter()          # Spatial    - edge-preserving spatial smoothing
temp_filter = rs.temporal_filter()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(config)
pc = rs.pointcloud()
points = rs.points()
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()

clipping_distance_in_meters = 1 #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale
intrinsics_depth = profile.get_stream(rs.stream.depth).as_video_stream_profile().get_intrinsics()
intrinsics_color = profile.get_stream(rs.stream.color).as_video_stream_profile().get_intrinsics()

pcd_depth_object = o3d.geometry.PointCloud()
try:
    # Wait for a coherent pair of frames: depth and color
    frames = pipeline.wait_for_frames()
    colorized = colorizer.process(frames)
    
    #aligning the rgb and the depth frames
    align = rs.align(rs.stream.color)
    frames = align.process(frames)  
    
    
    #Applying hole filter
    depth_frame = frames.get_depth_frame()
    color_frame = frames.get_color_frame()
    '''
    depth_frame = dec_filter.process(depth_frame)
    depth_frame = spat_filter.process(depth_frame)
    depth_frame = temp_filter.process(depth_frame)'''

    # Convert images to numpy arrays
    depth_image = np.asanyarray(depth_frame.get_data())
    color_image = np.asanyarray(color_frame.get_data())

    #Creating a binary mask of the same size as depth map for cropping
    mask_image = np.zeros_like(depth_image, dtype=np.uint8)

    #Obect ID for filtering the desired classes for detection
    desired_class_ids=[0]

    #Segmentation
    results=model.predict(source=color_image,stream=True)
    for person in results:
        indexes = [index for index, item in enumerate(person.boxes.cls) if item in desired_class_ids]
        mask=person.masks[indexes].xy #getting mask cordinates for each object detected in each frame
        #print(mask)
        for i in range(len(mask)): #iterating through each mask 
            
            act_mask = mask[i].tolist()  # Assuming the mask data is a list of coordinates
            
            # Convert the mask data into a list of contours
            mask_contours = [np.array(act_mask, dtype=np.int32)]

            cv2.fillPoly(mask_image, mask_contours, color=255) #Filling the mask_image with white pixels in place of mask coordinates
            #cv2.fillPoly(color_image, mask_contours, color=(0, 0, 255))

    # Apply colormap on depth image (image must be converted to 8-bit per pixel first)

    depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.05), cv2.COLORMAP_JET)
    depth_object = cv2.bitwise_and(depth_image, depth_image, mask=mask_image).

Here, depth_object is the cropped depth frame which is in numpy array format. I need to convert it into realsense depth frame format for extracting point clouds from it.

MartyG-RealSense, 2023-11-20:
Hi @sanjaiiv04  Converting numpy to rs2::frame remains an unsolved problem where there is no further advice available, unfortunately.

charlielito, 2024-04-18:
> depth_image

It's mind-blowing that 4 years later you haven't added support for this, unbelievable

mxshen, 2024-06-18:
I am waiting for the answer in 2024....

Ali-AliAli-Ali, 2024-11-29:
Following in 2024.

AviaAv, 2025-04-21:
Hey, 

While there isn't a built-in function in the SDK to convert an array back to a realsense frame, it can be done using a SW device - we can inject the array to a SW sensor and then get it as a frame.

We've just merged #13937 featuring an example of how to do that - see [numpy_to_frame.py](https://github.com/IntelRealSense/librealsense/blob/3a99bcf65e7871e8781fd253358fac570a4e8028/wrappers/python/examples/numpy_to_frame.py)

MartyG-RealSense, 2026-10-29:
Case closed due to RealSense implementing a method for addressing the issue in April 2025 - please see the comment https://github.com/realsenseai/librealsense/issues/5784#issuecomment-2817786435 above.

closed 2026-10-29 by MartyG-RealSense