realsenseai/librealsense #5786: Calibration file missing

evenlund, 2020-02-05:
The calibration file linked to in the readme.md (H_t265_d400.txt) is missing.

MartyG-RealSense, 2020-02-05:
It is a broken link.  You can get to it by using a link with a cfg extension instead of txt.

https://github.com/IntelRealSense/librealsense/blob/master/examples/tracking-and-depth/H_t265_d400.cfg

RealSenseCustomerSupport, 2020-04-05:
----------------------------------------------

Fixed in PR

https://github.com/IntelRealSense/librealsense/pull/6201

Thanks evenlund and MartyG!

closed 2020-04-24 by RealSenseCustomerSupport

egeozguroglu, 2024-07-28:
This file can't be accessed currently: https://github.com/IntelRealSense/librealsense/blob/master/examples/tracking-and-depth/H_t265_d400.cfg

MartyG-RealSense, 2024-07-28:
Hi @egeozguroglu  Support for the T265 camera model and online files / documentation for it were removed in librealsense SDK version 2.54.1 onwards.

I extracted the Tracking and Depth example from the source code for 2.53.1 and have attached it below as a zipped file.

[tracking-and-depth.zip](https://github.com/user-attachments/files/16402492/tracking-and-depth.zip)

The H_t265_d400.cfg file is a text file that simply contains the three lines below.

```
0.999968402 -0.006753626 -0.004188075 -0.015890727
-0.006685408 -0.999848172 0.016093893 0.028273059
-0.004296131 -0.016065384 -0.999861654 -0.009375589
```