realsenseai/librealsense #5790: PyRealSense2 PIP issue

neilyoung, 2020-02-05:
Today I was testing the PIP wheel of the 2.32.1 found here https://pypi.org/project/pyrealsense2/

It seems, that only Windows and Linux is supported currently. Would be great to have at least macOS and Raspbian too.

Anyway. I tested a script on my Ubuntu 18.04 with Python3.6. Before I installed the pyrealsense2 wheel:

```

decades@ubuntu:~/librealsense/build$ pip3 install pyrealsense2
Collecting pyrealsense2
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/28/48/b845fb2b5fad573fce1accb40ef5d1fcdf67d22bc5d57a2205b3dfb1e23d/pyrealsense2-2.32.1.1299-cp36-cp36m-manylinux1_x86_64.whl (9.1MB)
    100% |████████████████████████████████| 9.1MB 147kB/s 
Installing collected packages: pyrealsense2
Successfully installed pyrealsense2-2.32.1.1299
decades@ubuntu:~/librealsense/build$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> print(rs.__version__)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pyrealsense2' has no attribute '__version__'
>>> 
```

So there is a complain, that pyrealsens2 has no attribute '__version__'

I deinstalled the wheel and installed 2.32.1 python bindings from source. This is the result then:

```
decades@ubuntu:~/librealsense/build$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> print(rs.__version__)
2.32.1
```

OK, twice the same version, different result. Should that be the case?

MartyG-RealSense, 2020-02-05:
The most recent mention of pip install for MacOS that I know of was in March 2019.  At the time, the status was that it was something that was being looked at but was taking some time.  

https://github.com/IntelRealSense/librealsense/issues/3260#issuecomment-469361152

neilyoung, 2020-02-05:
@MartyG-RealSense Thanks, but macOS is now not my biggest concern. I'm wondering, why two more ore less identical versions can show a different behaviour. I know that the __version__ thing was added lately. Has it just been forgotten by the "Wheelies"?

MartyG-RealSense, 2020-02-05:
I'll have to defer that question to one of the Intel guys, as it is "outside of my wheel-house", as they say!

neilyoung, 2020-02-05:
Thanks :) Let 'em roll :)

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@neilyoung Is your request to add the **attribute 'version' for Linux/Windows?** Looking forward to your reply. Thanks!

neilyoung, 2020-02-14:
Exactly

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@neilyoung Thanks for the quick reply! Already feedback this to the engineering team. Any update will let you know. Thanks!

Nir-Az, 2024-05-26:
pyrealsense support both version types today at Linux and Windows.

`__version__`
and
`__full_version__`

Is this issue resolved?

sherrysun2018, 2024-06-04:
@neilyoung Could you please update if current implementation resolves your issue? Thanks!

neilyoung, 2024-06-04:
You are seriously asking me for a problem I probably had four (in words  f o u r) years ago with a product, which is stone dead since at least that long? We must be living in different worlds :)

closed 2024-06-04 by neilyoung