realsenseai/librealsense #5794: Intel RealSense depth map format

LMurphy99, 2020-02-05:
I'm trying to create my own point clouds from the realsense depth frames, as opposed to calculating one using the provided functionality. One key aspect I need to know is whether each depth pixel in the depth frame is distance relative to the center of one of the cameras, or whether the depth of each pixel is represented perpendicular to the XY plane.

Would really appreciate if anyone could let me know.
Cheers.

bmegli, 2020-02-06:
## Basics

See this:

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

## Details 

You are specifically interested in this function:

[`rs2_deproject_pixel_to_point`](https://github.com/IntelRealSense/librealsense/blob/d19829788008b8e000870895a068f0c43d58895a/include/librealsense2/rsutil.h#L67)

This is exactly what you want to do.

## Camera intrinsics and depth units

You will need to get depth units and camera intrinsics (focal points, principal points) and feed unprojection with this data.

See [example here](https://github.com/bmegli/realsense-network-hardware-video-encoder/blob/master/rnhve_hevc.cpp#L235) (camera intrinsics)

## Shortcuts

If you are using D435 (no distortion model) the unprojection simplifies to just few lines of code.

See independent example of unprojection working with D435 data [here](https://github.com/bmegli/hardware-depth-unprojector/blob/master/hdu.c#L54)

## Caveats

Camera intrinsics are expressed in pixel units and are valid only for the resolution you get them for.
Under realistic assumptions intrinsics scale with resolution (but this also depends on factors like changing FOV along changing resolution). 

On the other hand the distortion model doesn't scale with resolution (if your camera uses it).

If you don't know those details it's simplest to read the intrinsics and distortion model directly from the device for the resolution you use.

## Final notes

Good luck!

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Please let us know if we can close this ticket or there is any further question.
Thank you.

closed 2020-02-15 by LMurphy99