realsenseai/librealsense #5795: Standalone Application Deployment-Distribution Documentation

ltstein, 2020-02-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400/T26x | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Windows/Linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | All  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |Python, C++, Unity, OpenCV, OpenVX }                          | 
| Segment			|  Robot, AR/VR }                          | 

### Issue Description
Is there documentation on the proper way to package an application for distribution? I found an old reference on (application deployment)[https://software.intel.com/en-us/articles/application-engineering-guide-for-intel-realsense-application-deployment] but I haven't been able to find any recent information on how to go about including driver installers or necessary libraries for distribution of an application that utilizes D400 or T265 series cameras.

MartyG-RealSense, 2020-02-05:
I hope that the link below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/4277

ChrisProlls, 2020-12-17:
Same issue here.
We really want to avoid copying the SDK `realsense2.dll` because we assume that is for dev-only purpose.
Any ideas about how to deploy at large scale a standalone application without install anytime the SDK ?

MartyG-RealSense, 2020-12-17:
As this case is old I cannot handle it here, so could you re-post the question as a new case of your own by using the green **New Issue** button on the front page of the forum please?   Thanks!

https://github.com/IntelRealSense/librealsense/issues/

closed 2021-06-17 by RealSenseSupport