realsenseai/librealsense #5797: PCL Example Fail to Build...

jiapei100, 2020-02-06:
```console
[ 67%] Building CXX object examples/measure/CMakeFiles/rs-measure.dir/rs-measure.cpp.o
cd ....../librealsense/build_18.04/examples/measure && /usr/bin/c++  -DBUILD_EASYLOGGINGPP -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_LIBUVC_BACKEND -DUNICODE -I....../librealsense/realsense2 -I....../librealsense/src -I....../librealsense/examples/measure/rs-measure -I....../librealsense/examples/measure/.. -I....../librealsense/include  -I/opt/primesense/OpenNI/Include -I/usr/local/include/vtk -fPIC -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat-security -mssse3 -fopenmp -O3 -DNDEBUG   -std=gnu++11 -o CMakeFiles/rs-measure.dir/rs-measure.cpp.o -c ....../librealsense/examples/measure/rs-measure.cpp
In file included from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:2347:0,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::addSphere(const PointT&, double, double, double, double, const string&, int)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:581:24: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOn’
   actor->GetMapper ()->ImmediateModeRenderingOn ();
                        ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1490:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudGeometryHandler<PointT>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1524:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudColorHandler<PointT>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1600:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::addPolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&, int)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1699:35: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
     vtkIdType *cell = cell_array->WritePointer (vertices.size (), vertices.size () * (max_size_of_polygon + 1));
                                   ^~~~~~~~~~~~
                                   WeakPointers
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1876:28: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
   vtkIdType *cell = cells->WritePointer (verts.size (), verts.size () * (max_size_of_polygon + 1));
                            ^~~~~~~~~~~~
                            WeakPointers
In file included from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<ArrayType>& vtkSmartPointer<ArrayType>::operator=(const vtkSmartPointer<ArrayType>&) [with T = vtkRenderWindow]’:
/usr/local/include/pcl-1.10/pcl/visualization/interactor_style.h:364:18:   required from here
/usr/local/include/vtk/vtkSmartPointer.h:137:5: error: no matching function for call to ‘vtkSmartPointer<vtkRenderWindow>::operator=(vtkRenderWindow*)’
     this->vtkSmartPointerBase::operator=(r.GetPointer());
     ^~~~
In file included from /usr/local/include/vtk/vtkSmartPointer.h:26:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk/vtkSmartPointerBase.h:66:24: note: candidate: vtkSmartPointerBase& vtkSmartPointerBase::operator=(vtkObjectBase*)
   vtkSmartPointerBase& operator=(vtkObjectBase* r);
                        ^~~~~~~~
/usr/local/include/vtk/vtkSmartPointerBase.h:66:24: note:   no known conversion for argument 1 from ‘vtkRenderWindow*’ to ‘vtkObjectBase*’
/usr/local/include/vtk/vtkSmartPointerBase.h:67:24: note: candidate: vtkSmartPointerBase& vtkSmartPointerBase::operator=(const vtkSmartPointerBase&)
   vtkSmartPointerBase& operator=(const vtkSmartPointerBase& r);
                        ^~~~~~~~
/usr/local/include/vtk/vtkSmartPointerBase.h:67:24: note:   no known conversion for argument 1 from ‘vtkRenderWindow*’ to ‘const vtkSmartPointerBase&’
In file included from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk/vtkSmartPointer.h: In instantiation of ‘T* vtkSmartPointer<ArrayType>::GetPointer() const [with T = vtkRenderWindow]’:
/usr/local/include/vtk/vtkSmartPointer.h:137:44:   required from ‘vtkSmartPointer<ArrayType>& vtkSmartPointer<ArrayType>::operator=(const vtkSmartPointer<ArrayType>&) [with T = vtkRenderWindow]’
/usr/local/include/pcl-1.10/pcl/visualization/interactor_style.h:364:18:   required from here
/usr/local/include/vtk/vtkSmartPointer.h:181:43: error: invalid static_cast from type ‘vtkObjectBase* const’ to type ‘vtkRenderWindow*’
   T* GetPointer() const noexcept { return static_cast<T*>(this->Object); }
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

I am glad to hear the issue is resolved after upgrading the PCL library.
The rotation/flipping may be due to the use of different coordinate system,
Please refer to this similar case for ply export:
https://github.com/IntelRealSense/librealsense/issues/3589
Closing this issue for now. Please feel free to open a new issue for any other question.
Thank you.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

My apologies as the previous response was actually meant for #5635.
The compile error is related to the VTK.
Please check all the dependencies and minimum required versions for PCL are met. On windows for example:
http://pointclouds.org/documentation/tutorials/compiling_pcl_dependencies_windows.php
Please let us know if further help is needed.

Thank you.

jiapei100, 2020-02-14:
@RealSenseCustomerSupport 

I guess this bug may have something to do with **VTK** visualization, which is related to **PCL** visualization, which **librealsense** based upon. I'm using the **NEWEST** git of every package. Just wonder which version you are using to make your sample work ?

RealSenseCustomerSupport, 2020-02-20:
----------------------------------------------

I just use the same version as shown in the https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl
 PCL version 1.8.1
 VTK version 8.1.0
RealSense SDK build version 2.31.0
RealSense D435i Camera Firmware 5.12.2.100

    $ ldconfig -p | grep libpcl_visualization
     libpcl_visualization.so.1.8 (libc6,x86-64) => /usr/local/lib/libpcl_visualization.so.1.8
    $ ldconfig -p | grep libvtkCommonMisc
     libvtkCommonMisc-8.1.so.1 (libc6,x86-64) => /usr/local/lib/libvtkCommonMisc-8.1.so.1
    $ rs-enumerate-devices -s
    Device Name                  Serial Number      Firmware Version
    Intel RealSense D435I        843112073792        05.12.02.100

    $ ./librealsense-2.31.0/build/wrappers/pcl/pcl-color/rs-pcl-color

    Generate a Point Cloud? [y/n] y

    Generating PCD Point Cloud File...
    Captured_Frame1.pcd successfully generated.

Please let us know if further help is needed.
Thank you.

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Please let us know if further assistance is needed. Thank you.

closed 2020-03-06 by RealSenseCustomerSupport

jiapei100, 2020-03-12:
@RealSenseCustomerSupport 

I'm using March 11 2020's git.
- **PCL 1.10.0**
- **VTK 9.0.0**

```console
[ 53%] Building CXX object wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o
cd ....../librealsense/build_18.04/wrappers/pcl/pcl-color && /usr/bin/c++  -DBUILD_EASYLOGGINGPP -DBUILD_SHARED_LIBS -DDISABLE_LIBUSB_1_0 -DDISABLE_PCAP -DDISABLE_PNG -DDISABLE_VTK -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_CUDA -DRS2_USE_V4L2_BACKEND -DUNICODE -Dqh_QHpointer -I/usr/local/cuda/include -I....../librealsense/realsense2 -I....../librealsense/src -I....../librealsense/build_18.04 -I....../librealsense/include -isystem /usr/local/include/pcl-1.10 -isystem /usr/include/eigen3 -isystem /opt/primesense/OpenNI/Include -isystem /usr/local/include/opencv4  -I/opt/primesense/OpenNI/Include -I/usr/local/include/pcl-1.10 -I/usr/local/include/vtk-9.0 -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -fopenmp -O3 -DNDEBUG -fPIE   -std=gnu++14 -o CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o -c ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp
In file included from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:2347:0,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::addSphere(const PointT&, double, double, double, double, const string&, int)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:581:24: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOn’
   actor->GetMapper ()->ImmediateModeRenderingOn ();
                        ^~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1490:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudGeometryHandler<PointT>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1524:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePointCloud(const typename pcl::PointCloud<PointT>::ConstPtr&, const pcl::visualization::PointCloudColorHandler<PointT>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1600:38: error: ‘class vtkMapper’ has no member named ‘ImmediateModeRenderingOff’
   am_it->second.actor->GetMapper ()->ImmediateModeRenderingOff ();
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::addPolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&, int)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1699:35: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
     vtkIdType *cell = cell_array->WritePointer (vertices.size (), vertices.size () * (max_size_of_polygon + 1));
                                   ^~~~~~~~~~~~
                                   WeakPointers
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1876:28: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
   vtkIdType *cell = cells->WritePointer (verts.size (), verts.size () * (max_size_of_polygon + 1));
                            ^~~~~~~~~~~~
                            WeakPointers
In file included from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk-9.0/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<ArrayType>& vtkSmartPointer<ArrayType>::operator=(const vtkSmartPointer<ArrayType>&) [with T = vtkRenderWindow]’:
/usr/local/include/pcl-1.10/pcl/visualization/interactor_style.h:364:18:   required from here
[... cut ...]
wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/build.make:65: recipe for target 'wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o' failed
make[2]: *** [wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o] Error 1
make[2]: Leaving directory '....../librealsense/build_18.04'
CMakeFiles/Makefile2:1565: recipe for target 'wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all' failed
make[1]: *** [wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all] Error 2
make[1]: Leaving directory '....../librealsense/build_18.04'
Makefile:132: recipe for target 'all' failed
make: *** [all] Error 2
```

jiapei100, 2020-03-12:
Problem partly solved.... 
I have to make sure **VTK_RENDERING_BACKEND_OPENGL_VERSION** defined in file **pcl_config.h** is **bigger equal than 2**. In my case, I installed NVidia driver, here, **VTK_RENDERING_BACKEND_OPENGL_VERSION** should be **4**, which is bigger than 2.

Now, the ONLY leftover is as follows:

```console
[ 99%] Linking CXX executable rs-pcl
cd ....../librealsense/build_18.04/wrappers/pcl/pcl && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/rs-pcl.dir/link.txt --verbose=1
/usr/bin/c++  -I/opt/primesense/OpenNI/Include -I/usr/local/include/pcl-1.10 -I/usr/local/include/vtk-9.0 -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -fopenmp -O3 -DNDEBUG    -rdynamic CMakeFiles/rs-pcl.dir/rs-pcl.cpp.o  -o rs-pcl   -L/usr/local/lib  -L/opt/primesense/OpenNI/Tools  -Wl,-rpath,/usr/local/lib:....../librealsense/build_18.04:/opt/primesense/OpenNI/Tools: ../../../librealsense2.so.2.33.1 /usr/local/lib/libpcl_segmentation.so /usr/local/lib/libpcl_surface.so /usr/local/lib/libpcl_keypoints.so /usr/local/lib/libpcl_tracking.so /usr/local/lib/libpcl_recognition.so /usr/local/lib/libpcl_stereo.so /usr/local/lib/libpcl_cuda_features.so /usr/local/lib/libpcl_cuda_segmentation.so /usr/local/lib/libpcl_cuda_sample_consensus.so /usr/local/lib/libpcl_gpu_features.so /usr/local/lib/libpcl_gpu_segmentation.so -lboost_system -lboost_filesystem -lboost_date_time -lboost_iostreams -lboost_regex -lqhull -lOpenNI2 /usr/lib/x86_64-linux-gnu/libflann_cpp.so -lGL -lGLU /usr/local/lib/libglfw.so.3.4 /usr/local/lib/libpcl_ml.so /usr/local/lib/libpcl_registration.so /usr/local/lib/libpcl_features.so /usr/local/lib/libpcl_filters.so /usr/local/lib/libpcl_sample_consensus.so /usr/local/lib/libpcl_search.so /usr/local/lib/libpcl_kdtree.so /usr/local/lib/libpcl_io.so /usr/local/lib/libpcl_octree.so /usr/local/lib/libpcl_gpu_octree.so /usr/local/lib/libpcl_gpu_utils.so /usr/local/lib/libpcl_gpu_containers.so /usr/local/lib/libpcl_common.so 
In file included from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:2347:0,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::addPolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&, int)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1699:35: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
     vtkIdType *cell = cell_array->WritePointer (vertices.size (), vertices.size () * (max_size_of_polygon + 1));
                                   ^~~~~~~~~~~~
                                   WeakPointers
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp: In member function ‘bool pcl::visualization::PCLVisualizer::updatePolygonMesh(const typename pcl::PointCloud<PointT>::ConstPtr&, const std::vector<pcl::Vertices>&, const string&)’:
/usr/local/include/pcl-1.10/pcl/visualization/impl/pcl_visualizer.hpp:1876:28: error: ‘class vtkCellArray’ has no member named ‘WritePointer’; did you mean ‘WeakPointers’?
   vtkIdType *cell = cells->WritePointer (verts.size (), verts.size () * (max_size_of_polygon + 1));
                            ^~~~~~~~~~~~
                            WeakPointers
make[2]: Leaving directory '....../librealsense/build_18.04'
[ 99%] Built target rs-pcl
In file included from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk-9.0/vtkSmartPointer.h: In instantiation of ‘vtkSmartPointer<ArrayType>& vtkSmartPointer<ArrayType>::operator=(const vtkSmartPointer<ArrayType>&) [with T = vtkRenderWindow]’:
/usr/local/include/pcl-1.10/pcl/visualization/interactor_style.h:364:18:   required from here
/usr/local/include/vtk-9.0/vtkSmartPointer.h:137:5: error: no matching function for call to ‘vtkSmartPointer<vtkRenderWindow>::operator=(vtkRenderWindow*)’
     this->vtkSmartPointerBase::operator=(r.GetPointer());
     ^~~~
In file included from /usr/local/include/vtk-9.0/vtkSmartPointer.h:26:0,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_geometry_handlers.h:48,
                 from /usr/local/include/pcl-1.10/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/common/actor_map.h:41,
                 from /usr/local/include/pcl-1.10/pcl/visualization/pcl_visualizer.h:48,
                 from ....../librealsense/wrappers/pcl/pcl-color/rs-pcl-color.cpp:33:
/usr/local/include/vtk-9.0/vtkSmartPointerBase.h:66:24: note: candidate: vtkSmartPointerBase& vtkSmartPointerBase::operator=(vtkObjectBase*)
   vtkSmartPointerBase& operator=(vtkObjectBase* r);
                        ^~~~~~~~
/usr/local/include/vtk-9.0/vtkSmartPointerBase.h:66:24: note:   no known conversion for argument 1 from ‘vtkRenderWindow*’ to ‘vtkObjectBase*’
[... cut ...]
wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/build.make:65: recipe for target 'wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o' failed
make[2]: *** [wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/rs-pcl-color.cpp.o] Error 1
make[2]: Leaving directory '....../librealsense/build_18.04'
CMakeFiles/Makefile2:1565: recipe for target 'wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all' failed
make[1]: *** [wrappers/pcl/pcl-color/CMakeFiles/rs-pcl-color.dir/all] Error 2
make[1]: Leaving directory '....../librealsense/build_18.04'
Makefile:132: recipe for target 'all' failed
make: *** [all] Error 2
```