realsenseai/librealsense #5798: Errors of F200 on Raspberry Pi 4.

vocong25, 2020-02-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  F200  | 
| Firmware Version                   | 2.60.0.0 | 
| Operating System & Version |  Raspbian Buster| 
| Kernel Version (Linux Only)    | 4.19.75-v7l+                                        | 
| Platform                                 | Raspberry Pi|
| SDK Version                            |  legacy  
| Language                            |  {C/python }                          | 

### Issue Description

I have a problem when start a streaming camera of F200 on Raspberry Pi 4.
The device has been recognized and I can obtain RGB using Cheese on Raspbian Buster.
But when I started to stream frames through `pyrealsense` package, this error occurred: ` UVCIOC_CTRL_QUERY:UVC_SET_CUR error 2, No such file or directory`

Output here:

The left was from COLOR Stream and the right was from DEPTH Stream
[image]
 
did anybody face this problem? Please me some advices.

MartyG-RealSense, 2020-02-06:
Are you using the correct Pyrealsense for the legacy 1.12 SDK, please?  Pyrealsense2 is for the current RealSense SDK 2.0.  The similarly named but different Pyrealsense 2.x (e.g Pyrealsense 2.2) should be used with the legacy SDK.

The link below has a recently published guide to using an R200 (same generation as the F200) with the legacy SDK and the older Pyrealsense 2.2.  It may be adaptable for your F200.

https://idorobotics.com/2020/01/02/running-the-realsense-r200-camera-with-librealsense-and-pyrealsense-legacy-libraries/

vocong25, 2020-02-07:
Hi @MartyG-RealSense I've already installed and used legacy 1.12 SDK with pyrealsense (for legacy). 
I think the problem here is I haven't patched uvc module in raspbian buster (kernel 4.19) which causes errors. 
I have no idea how to patch it. I found some available scripts in the repo but all for other kernels/ubuntu version.
Any suggestions for me? Thanks in advance!

MartyG-RealSense, 2020-02-07:
The link below has a guide to patching uvcvideo on Raspian Buster.

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Patching_uvcvideo

vocong25, 2020-02-07:
Hi @MartyG-RealSense, It works like a charm.
But in some examples in cpp source and when using pyrealsense, I have a error: `rs.error: TemperatureControlLoop: opcodes do not match.`

[image]

MartyG-RealSense, 2020-02-07:
Apparently it may be a kernel compatibility issue between modern kernels and the legacy Librealsense.  A possible fix is provided in this link:

https://github.com/IntelRealSense/librealsense/pull/3929

vocong25, 2020-02-07:
Hi @MartyG-RealSense 
Wow. It's amazing! I've made F200 work on Raspberry Pi 4.

1. Solution #3929 works for me.
2. To build appropriate legacy SDK for ARM (Raspberry Pi 4 - Kernel 4.19), I used the combination: https://github.com/Maghoumi/librealsense and `pyrealsense` with some modifications in file `rs.h`.

Thank you so much!

vocong25, 2020-02-07:
Hi @MartyG-RealSense 
There are some problems with configuration.
Solution #3929 doesn't work :D

MartyG-RealSense, 2020-02-07:
I'll have to let @dorodnic advise on this one, as I do not have any experience with the legacy SDK, unfortunately.  

In the meantime, it is possible that you may be able to configure via ROS using the old **realsense_camera** package that is compatible with the old-generation cameras such as F200.

http://wiki.ros.org/realsense_camera

That would open up the possibility to do dynamic reconfiguring of a stream once it activates.

http://wiki.ros.org/realsense_camera#F200_Dynamically_Reconfigurable_Parameters

vocong25, 2020-02-12:
hi @MartyG-RealSense,

After few days trying to fix those bug, I found the way to overcome those problems.
`https://github.com/vocong25/librealsense_for_arm`
This is the step-by-step guideline which works on my Raspbian Buster - Kernel 4.19.97-v7l+ with Realsense F200.

Finally, thank you very much for your help!

MartyG-RealSense, 2020-02-12:
Great work!  Thanks so much for developing your solution and sharing it.    :)

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

@vocong25 Glad to hear the issue resolve. Thanks for sharing!

closed 2020-02-12 by RealSenseCustomerSupport