realsenseai/librealsense #5802: Question: Recording .bag frame-by-frame

MiguelKoch, 2020-02-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                         D400                               | 
| Firmware Version                   |                       5.11.6.250                         | 
| Operating System & Version |                  Win 10                                   | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 |                       PC                                     |
| SDK Version                           |                       2.22.0                                | 
| Language                               |                        python                             | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Question:

I want to synchronize the streams from an intel D415 camera and a ZED camera. I know how to record a stream into a .bag file and play it back, but that does not guarantee that both cameras are synced. Any ideas? 

The ZED API allows me to record frames into an .svo file in a "frame-by-frame" manner.
However, I don't know if it is possible to do the same for a .bag file.

One solution would be to use wait.for.frames and then retrieving the streams from the camera and saving them manually into .tif files, but I'm afraid that would not work for the desired framerate (25+).

MartyG-RealSense, 2020-02-06:
Data from RealSense and non-RealSense cameras can be aligned using camera metadata.  the link below discusses alignment between a RealSense and a ZED camera.

https://github.com/IntelRealSense/librealsense/issues/2186

MiguelKoch, 2020-02-06:
Thanks, I'll close this issue.

closed 2020-02-06 by MiguelKoch