realsenseai/librealsense #5804: Align depth frame to rgb image

VD2410, 2020-02-06:
I have a whole pipeline where a .Net frontend records the video from real sense and then saves both video and depth.

In the beginning, I was saving depth in the ".bag" file but because the file is too large and to better manage the storage I am now saving the depth in a compressed bits format ".dpt". If I say in short I can get the text file having depth value of every pixel. 

Now I know how to align the depth and RGB when I have the bag file but how to align the depth and RGB when I have depth values in a text file is my question. @dorodnic 

Thank you

ArjanVroegop, 2020-02-10:
Hi,

I think it's easier to align the frames before writing the depth data to your custom file format. Is this an option for you?

jb455, 2020-02-10:
You could use a `SoftwareDevice` ([example](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-4-software-dev)) to read your colour and depth data then run whatever filters you want on them, including align, but as @ArjanVroegop pointed out it would probably be easier to align before writing your files.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Hi VD2410. Do any of these recommendations help?

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-04-09 by VD2410