realsenseai/librealsense #5805: Feature Request: Allow the D435 IR Projector to be Always On.

doug-matthews, 2020-02-06:
### Issue Description
I was reading through Intel's white paper on projectors for the D4XX depth cameras and I came across their suggestion of using the projectors from multiple cameras. https://www.intelrealsense.com/wp-content/uploads/2019/03/WhitePaper_on_Projectors_for_RealSense_D4xx_1.0.pdf

In the article they say that the D435 pulses the projector in time with the frame capture, but that they disabled this for the test. I want to disable this feature for my application, but it appears this functionality is not possible through the SDK. 

Would it be possible to add this functionality to a future release? Thanks!

agrunnet, 2020-02-06:
Yes. Feature is either already out or will be very shortly. Let me check.

doug-matthews, 2020-02-19:
Hey @agrunnet just wondering if you have any updates yet on this feature? Thanks!

agrunnet, 2020-02-20:
Looks like it is being validated and should be released in about 2 weeks.

dorodnic, 2020-03-17:
I think you can try in on the development branch with latest firmware
Coming to next librealsense release

RealSenseCustomerSupport, 2020-04-03:
----------------------------------------------

Hi @doug-matthews Could you please try the latest LibRealSense release?

RealSenseCustomerSupport, 2020-04-14:
----------------------------------------------

Hi @doug-matthews Do you need further help?
If we don't hear from you in 7 days, we will clsoe this issue.

closed 2020-04-20 by RealSenseCustomerSupport