realsenseai/librealsense #5806: Cannot run 3 Cameras together on Ubuntu Linux (Crash)

jychong11, 2020-02-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 + D435 + T265 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.31 0 | 
| Language                            |  C++ | 

### Issue Description
Hello,

We are hitting a crashing bug when running 3 cameras together ( D415 + D435 + T265 ).
With 2 cameras ( any of the 2 ), it works with no issues.
But with 3 cameras, we get a "Device/Resource busy. rs2 error crash" when run.

As far as I can tell, the code dies when we hit the 2nd/3rd camera:

	// Start streaming with default configuration
	m_profile = m_pipe->start(cfg);

**Interestingly enough, the same code when run on Windows 10 works just fine!** 
We are hitting this issue when run on Ubuntu 18.04 LTS. We have built out 2.31.0 from source for Linux. We cannot use 2.31.2 because it will crash with a T265 ( please see the other bugs filed from other people recently for reference ). 2.31.0 works for 2 cameras ( T265 + D435/D415 ) but when we attach all 3 on Linux it crashes. Again, this issue does not happen on Windows 10.

Please let us know how to resolve this.

Thanks

zeentan, 2020-02-07:
Additional information:
Kernel: 5.3.0-28-generic

Seeing the following in dmesg
`
[  313.781293] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  313.831610] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  313.982078] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  314.032394] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  314.182960] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  314.233303] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  432.904972] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  432.956001] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  433.015711] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  433.081234] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
`

RealSenseSupport, 2020-07-25:
Hi,

Are you still experiencing this? We are able to successfully run all available streams simultaneously from all 3 devices using the latest librealsense (ver. 2.36.0) and f/w on both Windows 10 and Ubuntu 18.

FYI...if we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-08-14 by RealSenseSupport