realsenseai/librealsense #5807: Unity SDK Freezes On Dispose

BananaHemic, 2020-02-07:
I'm developing an AR application using the RealSense Unity SDK version 2.32.1. With Unity version 2019.2.13f1 on Windows 10, I'm experiencing consistent freezes in editor after exiting play mode. Digging deeper, it appears that m_pipeline.Dispose() in RsDevice is deadlocking. The Realsense device is the T265 firmware version 0.2.0.926.

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

Hi @BananaHemic

Thanks for the notification here.  Can you provide screenshots or logs that outline the deadlock you're were able to see during your debug?

BananaHemic, 2020-02-12:
Thanks for the reply, there was nothing unusual in the logs, everything operates normally until Unity exits play mode

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

Thanks @BananaHemic and you're only able to see this under Unity?  Have you attempted to use any other wrapper or use direct librealsense library calls?

BananaHemic, 2020-02-12:
I only tested with Unity

BlueprintBen, 2020-02-18:
I wanted to share some seemingly related data around my issues with changing device configurations on the fly (resolution, framerate, etc), which seem to center on a huge hang in the RsDevice.OnDisable() function:
[image]
Given that this call takes 20 seconds I'd agree that there's a serious timing concern to address. This is using `Multithreaded` mode, and stopping the camera via `DestroyImmediate(RsDevice.gameObject)`. Haven't used other wrappers/library platforms.

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

Thanks @BlueprintBen and @BananaHemic for the feedback and information.

Carlo-Gezapp, 2020-10-12:
Any new info? Still same problem with T265 0.2.0.951 in unity 2019.1.8! Thanks

BlueprintBen, 2021-01-01:
Can confirm that this issue still manifests with D435i's updated to the latest firmware and Unity SDK. Since this causes severe unresponsiveness while changing stream configurations, our continuing plan is to ignore IMU support for RealSense devices in our product and advertising that functionality solely for Azure Kinect & ZED cameras instead.

closed 2021-04-20 by RealSenseSupport

BlueprintBen, 2021-05-03:
Continuing to experience this issue using RealSense SDK 2.44, resulting in 20 second any time the active profile is changed 😒
[image]
Using RealSense D435i - FW 05.12.12.100