realsenseai/librealsense #5808: Announcement: Intel updates RealSense device selection guide to include L515 lidar camera

MartyG-RealSense, 2020-02-07:
Hi everyone,

Intel have updated their 'Which Intel RealSense device is right for you?' comparison guide to include the forthcoming new RealSense L515 lidar camera.

https://www.intelrealsense.com/which-device-is-right-for-you

ShettyHarapanahalli, 2020-02-11:
Do we have the accuracy for each device at the minimum depth range (Z) assuming we operate at night to avoid lighting condition variations?

MartyG-RealSense, 2020-02-11:
Although it is possible to describe accuracy with a single line, in truth it is more complex than that and depends on a range of factors.

The link below describes in detail the accuracy of the 400 Series cameras.

https://github.com/IntelRealSense/librealsense/issues/999#issuecomment-366407953

The official L515 information page's FAQ gives the following information for the L515's depth accuracy.

Depth error average at 1m distance from camera is <5mm
Depth error std deviation at 1m distance from camera is 2.5mm*
Depth error average at 9m distance from camera is <14mm*
Depth error st deviation at 9m distance from camera is 15.5mm*
*at 95% reflectivity

The SR300 / SR305 has a depth unit value of 1/32 of a mm, smaller than the current minimum depth unit achievable on the 400 Series.  This means that the SR300 / SR305 may have better accuracy than the 400 Series at very close range to an object, but otherwise a 400 Series or L515 would likely give better results.  The accuracy of the 400 Series can be < 1% , whilst the older SR300 technology's error percentage is around 5%.

closed 2020-02-13 by MartyG-RealSense