realsenseai/librealsense #5809: Proprietary license in src/win7/drivers/IntelRealSense_D400_series_win7.inf

jirislaby, 2020-02-07:
Could you lift the wording in the license here:
https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/src/win7/drivers/IntelRealSense_D400_series_win7.inf#L5-L10
?

It looks like very strict WRT to distribution in a linux distribution.

dorodnic, 2020-02-08:
Hi @jirislaby 
Yes, sure, will update it. Previously Windows 7 specific drivers used to be distributed under proprietary license, but being part of the SDK now, should be updated to Apache.

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

Hi @jirislaby, the license is updated in the latest SDK release build v.2.33.1.
Thank you.

closed 2020-03-25 by RealSenseCustomerSupport