realsenseai/librealsense #5811: Missing link

neilyoung, 2020-02-07:
A couple of days ago a post went through this list, which covered the question: How can I create my own CPP RealSense project? I can't find that anymore. I think it was initially of end of June last year. The question was, what do I need to do in order to create my own project in CPP? The question was answered by Jim as far as I remember and refreshed these days by Marty. Does anybody remember that thread and would be so kind as to post the link here?

Up to now whenever I create a new CPP project I copy an existing one in the tree of the SDK and manipulate the global CMakeList.txt or so. I would like to change that, but my CMAKE knowledge is not that good.

TIA

MartyG-RealSense, 2020-02-07:
It was this link:

https://github.com/IntelRealSense/librealsense/issues/4277

neilyoung, 2020-02-07:
Exactly. Many thanks. :)

closed 2020-02-07 by neilyoung