realsenseai/librealsense #5812: How to get pointcloud2 by realsense d435 using Matlab

sonumathur, 2020-02-07:
| Camera Model                       | D435 |  
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |             
| Language                            |  Matlab    | 
|
### Issue Description

I want to process .xyz file from D435 in Matlab, but it Require PointCloud2 object as given in https://in.mathworks.com/help/ros/ref/pointcloud2.html#bu3uilg

I can save the using realsense D435 as .ply but it is not pointCloud2 as required in Matlab.

There are many methods for converting the ply to .xyz but my moto is to direct access of incoming data from realsense to preserve the shape of object......

This method suggested by you is also working but I think there a loss feature by conversion.

So my Aim here is to access the realsense data to pointcloud2.

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

I think this project will help you.
https://github.com/IntelRealSense/realsense-ros

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L805

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@sonumathur Did you get chance to try? Looking forward to your update. Thanks!

sonumathur, 2020-02-27:
I have no knowledge about how to work with ROS as replied by @RealSenseCustomerSupport .

Till not able to generate pointcloud2 object using D435 in Matlab.
If there is any method for Matlab then please provide.....

Thanks

RealSenseCustomerSupport, 2020-03-02:
----------------------------------------------

Sorry, PointCloud2 is a kind of ROS message.
so far, get pointcloud2 object via ROS is the easiest way.

dorodnic, 2020-03-17:
I think there is some confusion,
@sonumathur - have you reviewed [pointcloud_example.m](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/matlab/pointcloud_example.m). The team is not very experienced with Matlab, so converting to pointCloud2 is more of a general Matlab question, but you should be able to fetch all the raw data using the provided example

sonumathur, 2020-03-18:
Thanks for your reply @dorodnic 

Actually I tried many solutions but still not get the pointcloud2 object.
My main aim is to generate a .xyz data of object under D435. and I think that is generated under pointcloud2.
If there is any way to generate pointcloud2 data using realsense in other Language, Please Provide.

Thanks.

RealSenseSupport, 2020-07-02:
@sonumathur Sorry for late reply! Is this still problem for you? Looking forward to your reply. Thanks!

sonumathur, 2020-07-02:
Till today there is no update in this regard...........

RealSenseSupport, 2020-08-04:
according to the page https://www.mathworks.com/help/ros/ref/pointcloud2.html
> The PointCloud2 object is an implementation of the sensor_msgs/PointCloud2 message type in ROS.

You can receive the PointCloud2 object from the ROS wrapper.

Or, try the script in this page to convert xyz data to pointcloud2 object in Matlab.
https://www.mathworks.com/matlabcentral/answers/395620-ros-create-pointcloud2-from-matlab-pointcloud

RealSenseSupport, 2020-09-04:
@sonumathur Did you get chance to try? Looking forward to your update. Thanks!

sonumathur, 2020-09-05:
@RealSenseSupport  I try many times but not able to get desired result.
Under ROS I try following:
roslaunch realsense_ros_camera rs_rgbd.launch

rosrun pcl_ros pointcloud_to_pcd input:=/sensor_msgs/PointCloud2

But there is no output Pointcloud file from this topic /sensor_msgs/PointCloud2

RealSenseSupport, 2020-10-16:
the topic name is "/camera/depth_registered/points" by default.
the type of this topic is "sensor_msgs/PointCloud2"

RealSenseSupport, 2020-11-06:
@sonumathur Any other questions about this? Looking forward to your update. Thanks!

closed 2020-11-06 by sonumathur