realsenseai/librealsense #5813: D435 RGB camera does not work after firmware upgrade

RyanHCCSI, 2020-02-07:
Property|Value
------|-----
Name|Intel RealSense D435
Camera Model|D400
Firmware Version|05.12.02.100
Operating System & Version|Windows 10 Pro (10.0.17763 Build 17763)
Platform|PC
SDK Version|2.32.1
Language| N/A

### Issue Description
I installed the RealSense SDK v2.32.1 for the first time and everything seemed to be running fine. I upgraded from an old firmware, I did not note the version but I believe it was factory, to the latest 5.12.02.100. When starting the Intel RealSense Viewer I receive the message below
> UNKNOWN in rs2_get_option_range(options:0000022B6B661780, option:Auto Exposure Priority, min:-nan, max:0, step:7.77721e-43, def:2.62831e+36):
> The device does not support camera settings such as zoom, pan, aperture adjustment, or shutter speed.

The Stereo Module seems to work, but when attempting to enable the RGB Camera the following RealSense error calling message is raised:
> Invalid Value in rs2_open_multiple(sensor:0000022B6B661780, profiles:0000022B7C5E9558, count:1):
>
> Failed to resolve the request: 
> 	Format: RGB8, width: 640, height: 480
>
> Into:
> 	Formats: 
> 	 YUYV

### Troubleshooting Steps Taken
* Different RGB Camera options
* A 05.11 series firmware using DFU tool
* Camera Reset through RealSense Viewer 2.32.1
* Physically unplugged camera
* RealSense Viewer 2.31 installation
* RealSense upgrade to Viewer 2.32.1 again
* RealSense firmware upgrade to 05.12.02.100 again
* More Camera resets

RealSenseCustomerSupport, 2020-02-10:
----------------------------------------------

Hi RyanHCCSI,
-please try to use viewer embeded FW update tool to burn the recommended FW again to see if help

you can find that under the More tab at the upper right side of viewer

More->Install recommended FW ( 5.12.02.100)
you can also click the info tab to check the FW version

-when plug the camera to PC
under device manager, could you see both Camera D435 with RGB module depth
and Camera D435 with RGB module RGB are enumerated?

if yes, you can uninstall all D435 drivers (include hidden ones) then plug the D435 again
Device manager->View->show hidden devices
under Camera class, uninstall all D435 related drivers

-what is the 12 digits Serial number of that D435 camera?
yo can find it by click the Info tab

best

Attachment(s):
instawll recommend FW.pptx - https://realsensesupport.intel.com/attachments/token/N3eWtkZu0DOqKTu4porJY2PVY/?name=instawll+recommend+FW.pptx

RyanHCCSI, 2020-02-11:
The uninstall and reinstall seemed to do the trick. Thank you for your help.

closed 2020-02-11 by RyanHCCSI