realsenseai/librealsense #5815: Errors logged whenever realsense-viewer is started

Ezward, 2020-02-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 5.3.0-28-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.32.1 | 
| Language                            | realsense-viewer | 
| Segment			| Robot-donkeycar | 

### Issue Description
realsense-viewer logs errors whenever it starts up.  The viewer works, but I don't understand if it is somehow impaired due to these errors;

``` 08/02 11:53:25,057 WARNING [139685282928384] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 08/02 11:53:25,097 WARNING [139685282928384] (backend-v4l2.cpp:1208) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 08/02 11:53:25,114 ERROR [139685282928384] (types.h:304) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid argument
 08/02 11:53:25,114 WARNING [139685282928384] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 08/02 11:53:25,114 ERROR [139685282928384] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 08/02 11:53:25,114 WARNING [139685282928384] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure
```

Those are the errors at startup.  These errors are logged when I turn on an off the depth, rgb and imu streams.

``` 08/02 11:52:45,175 ERROR [139830684108544] (types.h:304) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
 08/02 11:52:45,175 WARNING [139830684108544] (sensor.cpp:630) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure
 08/02 11:52:50,550 WARNING [139830512023296] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/02 11:52:52,051 WARNING [139830503630592] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/02 11:52:53,095 WARNING [139829975185152] (sensor.cpp:968) HID timestamp not found, switching to Host timestamps.
```

dorodnic, 2020-02-08:
Hi @Ezward 
Most of these are due to lacking kernel patches (specifically all of the warnings). Not sure what is the nature of the errors. 
For D400 cameras on kernel 5.3 not having kernel patches have some small negative effects, like not having most accurate time-stamping and synchronization. 
We are hard at work adding support for 5.x kernel versions. If you follow Linux installation instructions and apply scripts from the development branch, you should be able to get everything working. Also, `librealsense2-dkms` package was either already published or will be shortly for the 5.3 version.

Ezward, 2020-02-09:
I'll wait for the 5.3 kernel support.

closed 2020-02-09 by Ezward