realsenseai/librealsense #5819: Set D435i to sleep mode.

deepak-1530, 2020-02-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   |2.22.0 | 
| Operating System & Version |  Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (4.15.0-76-generic)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 

### Issue Description
Is there a way to put D435i camera on sleep more or low power mode when connected to USB port?

ev-mp, 2020-02-10:
@deepak-1530 , the camera's power management is governed by the USB protocol policies implemented in the controller/kernel.
 On most OS-es the transition from high to low-power state will occur automatically within 3 seconds* of idle operation (no streaming and no controls sent/received). 
*The threshold can be reconfigured for Linux, probably for the other OS-es as well.

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

Hi @deepak-1530, Do you have further question?
If we don't hear from you in 7 days, this issue will be closed.

deepak-1530, 2020-02-15:
No

closed 2020-02-15 by deepak-1530